e2nIEE/pandapower #1: The array index will not be correctly found when pvpq array is not sorted

chengts95, 2017-02-04:
Check This [line](https://github.com/panda-power/pandapower/blob/develop/pandapower/pypower_extensions/newtonpf.py#L64)
`pvpq_lookup = zeros(pvpq[-1]+1, dtype=int)`, I think the `pvpq[-1]`  may not always be the largest index of the pvpq array, it should be like `pvpq_lookup = zeros(pvpq).max() , dtype=int)` to ensure the largest index will be always found. This can help to compatible with common pypower case format.

lthurner, 2017-02-06:
The functions in pandapower/pypower_extension are for now only recommended to use within pandapower, not with casefiles that were not generated by pandapower. 

We want to make the numba accelerations available for pypower, so that they work for all pypower casefiles. There are probably some issues to resolve before this generally works, and this is one of them, so thanks for pointing it out.

chengts95, 2017-02-07:
Thanks  for the  reply, I was calculating a customized power systemwhich had been created in pypower before in pandapower. In fact I used the `converter.fromppc` at first, but the conversion was a little inaccurate due to the differences in the transformer's model(seems to be another issue). that is why I modified some functions in `pypower_extension` to make the pandapower's runpf directly compatible with pypower cases to verify the results.

jhmenke, 2017-02-07:
If the transformer caused differences after conversion you may look at the trafo_model parameter of your transformers. Maybe changing the type from "t" to "pi" or vice versa is enough to fix your problem.

http://www.uni-kassel.de/eecs/fileadmin/datas/fb16/Fachgebiete/energiemanagement/Software/pandapower-doc/elements/trafo.html#electric-model

commit 84bb12c6 (2017-10-11): bug fixes: (from_ppc - max_i_ka with sqrt(3)) & (test_case9_conversion max_loading_percent)
-> correct line.max_i_ka in power_system_test_cases
close #1
commit cf4bd4b8 (2017-10-11): improve documentation
close #1
commit 1cc7f46d (2017-10-11): bug fixes: (from_ppc - max_i_ka with sqrt(3)) & (test_case9_conversion max_loading_percent)
-> correct line.max_i_ka in power_system_test_cases
close #1
commit ed751682 (2017-10-11): improve documentation
close #1
closed 2017-05-02 by lthurner