e2nIEE/pandapower #2: tests for topology module

panda-power, 2017-02-15:
There are no tests for the topology module. Tests could be derived from the examples in the documentation

lthurner, 2017-02-15:
test coverage rate is now available on coveralls: https://coveralls.io/github/lthurner/pandapower?branch=master and is currently at 74%

The main holes in the coverage seem to be:
- coveralls does not seem to work with JIT (e.g. pypower_extension/create_J is marked as not tested even though it definitely should be)
- incomplete/missing tests for the toolbox
- incomplete tests for create functions
- incomplete tests for topology
- missing tests for diagnostic_reports (is this necessary? could also be excluded from coverage)

lthurner, 2017-05-04:
coverage was moved to codecov, since coveralls was not working properly: https://codecov.io/github/lthurner/pandapower?branch=develop

Current coverage rate is at 87%

closed 2017-05-17 by lthurner