e2nIEE/pandapower #4: save pandapower net as JSON file

lthurner, 2017-02-15:
The pickle files can only be read from python, so it would be helpful to have a possibility to save pandapower networks in an universally readable format such as JSON.

jhmenke, 2017-02-15:
here is sample code to convert to json.

```
def to_json(net, filename):
    json_string = "{"
    for k in sorted(net.keys()):
        if k[:6] == "_empty":
            continue
        if k in ["std_types", "_mpc_last_cycle"]:
            continue
        if isinstance(net[k], pd.DataFrame):
            json_string += '"%s":%s,' % (k, net[k].to_json())
        elif isinstance(net[k], bool):
            json_string += '"%s":%s,' % (k, "true" if net[k] else "false")
        elif isinstance(net[k], str):
            json_string += '"%s":"%s",' % (k, net[k])
        elif isinstance(net[k], numbers.Number):
            json_string += '"%s":%s,' % (k, net[k])
        elif net[k] is None:
            json_string += '"%s":null,' % net[k]
        else:
            logging.error("could not detect type of %s: %s" % (k, net[k]))
    with open(filename, "w") as text_file:
        text_file.write(json_string[:-1] + "}\n")
```

jhmenke, 2017-02-17:
I also implemented from_json. Will test and commit today.

jhmenke, 2017-02-23:
experimental version is pushed. 
at the moment the index of dataframes is sorted in ascending order. it does not seem possible to retain the original "unsorted" pandapower indices.

jhmenke, 2017-03-21:
Added, tested and will be released in pandapower 1.2

closed 2017-03-21 by jhmenke