e2nIEE/pandapower #6: grids without slack nodes

FlorianShepherd, 2017-02-28:
Pypower allows grids withouth slack nodes and changes the first PV-bus to a slack node if none is available.
I think it would be better if a warning / error is raised instead and the toolbox should get a function to change a PV node to a slack node. 

See bustypes.py:
`
    # pick a new reference bus if for some reason there is none (may have been
    # shut down)
    if len(ref) == 0:
        # bugfix Pypower: must be an numpy array!
        ref = zeros(1, dtype=int)
        ref[0] = pv[0]      # use the first PV bus
        pv = pv[1:]      # take it off PV list
`

lthurner, 2017-03-16:
With the new connectivity check, we can now detect islands. I would suggest the following behaviour:
- if an island has neither gen nor ext_grid, set it out of service (now already done by connectivity check)
- if one generator but no ext_grid is in one island, treat the generator as a slack
- if there is at least one ext_grid in an island, proceed as usual
- if no ext_grid and multiple generators are in one island, throw an error

FlorianShepherd, 2017-03-16:
Sounds valid.

One remark: In case 2 "if one generator but no ext_grid is in one island, treat the generator as a slack" and the generator limits are set too tight, the power flow won't converge. This could lead to problems which are hard to find. On the other hand this also sounds like a quite unusual case...

FlorianShepherd, 2017-03-20:
Fixed. We agreed on the following:

- If an island is present (=connected buses without a slack/ext_grid), no matter if it contains gens, the isolated buses will be set "out of service"
- If no slack / ext_grid is available in the whole net, an error is raised

pandapower.diagnostic can be used to identify unsupplied buses with gens. These can be manually converted to ext_grid if desired. Otherwise the behaviour of the power flow algorithm would be implicit.

closed 2017-03-20 by FlorianShepherd