e2nIEE/pandapower #10: Couldn't import dot_parser, loading of dot files will not be possible.

vitordsbatista, 2017-03-15:
Hello, I found a bug in your package.
When I tried to import an MATPOWER case the following error occurred:

```
Couldn't import dot_parser, loading of dot files will not be possible.
Traceback (most recent call last):
  File "<stdin>", line 32, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pandapower/converter/matpower/from_mpc.py", line 49, in from_mpc
    ppc = _mpc2ppc(mpc_file, casename_mpc_file)
  File "/usr/local/lib/python2.7/dist-packages/pandapower/converter/matpower/from_mpc.py", line 63, in _mpc2ppc
    _adjust_ppc_indices(ppc)
  File "/usr/local/lib/python2.7/dist-packages/pandapower/converter/matpower/from_mpc.py", line 82, in _adjust_ppc_indices
    ppc["gen"][:, 0] -= 1
IndexError: too many indices
```

This error occurred because sometimes ppc["gen"] could be an array, not a matrix. So, I reshaped the array to transform it in a single line matrix by adding the following code before it.

```
def _adjust_ppc_indices(ppc):
    # adjust indices of ppc, since ppc must start at 0 rather than 1 (matlab)
    ppc["bus"][:, 0] -= 1
    ppc["branch"][:, 0] -= 1
    ppc["branch"][:, 1] -= 1
    #Start of changes by Vitor Batista
    """
    shp = ppc['gen'].shape
    if len(shp) == 1:
        ppc['gen'] = np.reshape(
            ppc['gen'], [1, shp[0]])
    """
    #End of changes by Vitor Batista
    ppc["gen"][:, 0] -= 1
```

Probably there is a faster way to solve it, but it fixed my problem.

lthurner, 2017-03-16:
Hi, thanks for the feedback, makes sense to check for one-dimensional gen arrays. Do you want to submit your changes as a pull request?

vitordsbatista, 2017-03-20:
Yes. However, first I will find a faster way to modify it.

lthurner, 2017-03-20:
importing network data is usually done exactly once and not repeatedly, so I don't think performance is a crucial issue here. The rest of the converter is also not optimized much for speed, so in my opinion the solution you posted is absolutely fine.

vitordsbatista, 2017-03-20:
I know, but I work with genetic algorithm, I must call the power flow at least 10000 times, maybe more, so I need the power flow be fast.

lthurner, 2017-03-20:
I agree the power flow has to be fast, and we optimized the power flow to be as fast as possible. But the conversion from matpower is not part of the power flow, so I don't understand what workflow would require you to call the conversion 1000 times? I would think you would import from matpower to pandapower once and then do 1000 power flows just in pandapower?

vitordsbatista, 2017-03-23:
You're right, I only will convert some matpower case once. Sorry for that.
Just one more question. Do you think it's necessary check the shape of another variables form the dict, like ppc["bus"] and ppc["branch"]?

lthurner, 2017-03-23:
there is probably going to be very few cases with one bus or one branch, but it can't hurt, so if you are already at: why not :)

lthurner, 2017-04-03:
are you still doing this or shall we add the fix ourselves?

vitordsbatista, 2017-04-14:
Sorry to not answer before, many things to do. I'm not developing the faster way, as you say the program will only import the matcases once. I thought you'd better fix it by yourselves.
Thank you for your understanding

lthurner, 2017-04-19:
Fixed:

https://github.com/lthurner/pandapower/blob/develop/pandapower/converter/matpower/from_mpc.py#L74

closed 2017-04-19 by lthurner