e2nIEE/pandapower #16: Storing Prices after OPF

evanhaldane, 2017-03-29:
It looks like in the `_copy_results_ppci_to_ppc` [function](https://github.com/lthurner/pandapower/blob/9c5522c0d2723624e07b9103fae50b73e93dd44f/pandapower/results.py#L92), you only take the first 13 columns of the `result['bus']` matrix. This means that the stored results don't include the Lagrange multipliers (in other words, the marginal prices).

Would it be possible to include those in the results that are saved after an OPF?

lthurner, 2017-03-30:
good point. The number of columns n_cols is taken from ppc, which does not include the lagrange columns. It should be taken from result instead.

Fixing this would lead to having the marginal prices in the internal ppc. Then it should be straightforward to also include them in the res_bus table. Do you want to take a shot at it?

evanhaldane, 2017-03-30:
I'm new to this package, so my main concern is whether that would break anything else. 

Nevertheless I'll give it a shot and submit a PR.

lthurner, 2017-03-30:
We have a test suite with over 200 unit tests that cover around 90% of the code base (see here: http://pandapower.readthedocs.io/en/latest/about/tests.html)

Of course its not guaranteed that everything works if all tests pass, but its a pretty good indicator. And if you have any questions, you can always ask!

lthurner, 2017-04-03:
request is merged, thanks evan!

closed 2017-04-03 by lthurner