e2nIEE/pandapower #18: simple_plot fails when bus_geodata or line_geodata are deleted

jkrstulo, 2017-03-31:
this code fails due to KeyError:

from pandapower.networks import mv_oberrhein
import pandapower.plotting import simple_plot

if __name__ == '__main__':

    net = mv_oberrhein()

    del net.bus_geodata 
    del net.line_geodata

    ppplot.simple_plot(net)

existance of bus_geodata and line_geodata is checked with assumption that keys line_geodata and bus_geodata exist:
if len(net.line_geodata) == 0 and len(net.bus_geodata) == 0:
        logger.warning("No or insufficient geodata available --> Creating artificial coordinates." +
                       " This may take some time")
        create_generic_coordinates(net, respect_switches=respect_switches)

lthurner, 2017-04-03:
I think this is an old code, where the geodata is deleted in the wrong way. pandapower always assumes that all tables exist, the same goes for element tables (if you delete any element table like that, the power flow would also give an error because it can't find certain tables). The correct way to delete all geodata (or all elements of any other table) would be:

net.bus_geodata.drop(net.bus_geodata.index, inplace=True)

jhmenke, 2017-04-03:
Since bus_geodata and line_geodata are created by default, there is the question whether both dataframes should even be deleted from a pandapower network.

Instead, you should use

    net.bus_geodata.drop(net.bus_geodata.index, inplace=True)
    net.line_geodata.drop(net.line_geodata.index, inplace=True)

to clear the data from the network.

closed 2017-04-03 by jhmenke

jhmenke, 2017-04-03:
Commit 9ae5b8f42f69ae3 also changes the behaviour in the plotting tutorial.

jkrstulo, 2017-04-04:
I am sorry for this...I came into this issue because I followed the tutorial...and then I tested on the same code for plotly plots. It makes much more sense to use drop for "safe" delete and assuming all the tables are present.
thanks both of you and sorry for taking your time to respond.