e2nIEE/pandapower #21: Rearrangment of pandapower file structure

FlorianShepherd, 2017-04-05:
since we're discussing the standardization of the idx files and decided to integrate the pypower files (since the current PIP version is not compatible with numpy 1.12), It may be better to reorder the pandapower files in the main directory to increase readability. The current subfolders (converter, plotting, networks...) are extensions to pandapower, all remaining files are somehow classified as "framework". When I look at pp/tests, there is already some structure (api for example tests the toolbox, loadflow the powerflow and so on). Maybe we should try to keep that consistent. Here is a suggestion:

-   converter, estimation, networks, plotting, shortcircuit, topology stay as they are
-     powerflow will be added
-     opf will be added
-     core / framework / tbd will be added (contains build files, pd2ppc, idx files, results)
-     pypower_extensions will be deleted and the files moved to powerflow / opf

I'm undecided about api in tests. Isn't an API a programming interface for other services to use the functionality of the software?

jhmenke, 2017-04-06:
>I'm undecided about api in tests. Isn't an API a programming interface for other services to use the functionality of the software?

Yes. "Misc" or "Core" would probably be more fitting.

lthurner, 2017-04-06:
Yes I agree. I think the structure you propose is fine, but it will be difficult to seperate powerflow and opf completely, since opf uses a lot of functions from powerflow also. So the opf folder would then only be "all opf functions that are not already in powerflow"

FlorianShepherd, 2017-04-06:
You're right about the powerflow / opf functions. But I still think it would be "understandable" to seperate it like this. An alternative would be to put every powerflow function into "core", but I'd prefer the powerflow folder.

P.S.: I think this would be a great module name: "all opf functions that are not already in powerflow"
import pandapower.allopffunctionsthatarenotalreadyinpoweflow 💃 ;)

FlorianShepherd, 2017-04-06:
At first I'll integrate the remaining newton-PF functions from Pypower into pypower_extensions to solve the numpy 1.12 issue. This would be:
pypower.:
idx_gen
idx_bus
idx_brch
makeSbus
dSbus_dV (no numba)

What about gausspf and fdpf? Shall this be updated as well? Then these would be integrated as well:
makeB
ppoption
gausspf
fdpf

Since we do not officially maintain these, I would leave those with pypower. 

As a second step I'd change the folder structure. To avoid merge conflicts, could you, Jakov ( @jkrstulo ), please tell me which functions you changed for the ZIP loads?

jkrstulo, 2017-04-06:
@FlorianShepherd heh I needed to change plenty of them for zip loads, here's the list:
+ auxiliary
+ build_bus
+ create
+ run
+ run_newton_raphson_pf
+ create_py

pypower_extensions/
+ dSbus_dV
+ newtonpf
+ pfsoln

converter/pypower
+ to_ppc

maybe you would prefer that I first finish with this (give me 1-2 days, I don't have too much work left, but I will be a bit busy with other things), push it and then you change folders?

FlorianShepherd, 2017-04-06:
@jkrstulo Thank you Jakov for the list! Yes of course, don't worry and take as much time as you need. The restructuring is not crucial in terms of time.

FlorianShepherd, 2017-05-04:
Filestructure Changed. See commit https://github.com/lthurner/pandapower/commit/a857c84a52f717b9f539600b076356c33d5eab57

files relevant for powerflow are moved to pf, similary with optimal power flow files.

closed 2017-05-04 by FlorianShepherd