e2nIEE/pandapower #22: shunt model: input parameters and taps

lthurner, 2017-04-06:
In #20 we discussed ZIP load parameters and decided to include the shunt-impedance as a a relative value instead of explicit power parameters pz_kw / qz_kvar. Does this also make sense for the shunt model, e.g. to make two parameters rated power and an active power loss percentage instead of pz_kw / qz_kvar?

Also, it would maybe be helpful to have a tap model for shunts as well, so that a rated power for each step can be defined and the shunt power can be increased / decreased stepwise?

jkrstulo, 2017-04-13:
> Does this also make sense for the shunt model, e.g. to make two parameters rated power and an active power loss percentage instead of pz_kw / qz_kvar?

I don't get what do you mean by this...it would be percentage of what?

> Also, it would maybe be helpful to have a tap model for shunts as well, so that a rated power for each step can be defined and the shunt power can be increased / decreased stepwise?

this makes a lot of sense definitely...it could be eventually very useful for some Volt/var studies...and maybe distribution OPFs or similar problems..

lthurner, 2017-04-13:
> I don't get what do you mean by this...it would be percentage of what?

I am not sure, but some relative value either to a rated power or the reactive power? PowerFactory for example takes a loss factor tan(alpha) as input for active power losses. I would have to check other software to see if this is a common parameter

lthurner, 2017-04-19:
there should also be a vn_kv parameter instead of taking vn_kv from the bus table since the element could have a different rated voltage than the bus

lthurner, 2017-05-04:
I have added the columns "vn_kv" and "step" to the shunt table in commit 7340c0300ba79e58468a171f5f5bd15762a8824e

"vn_kv" allows to specify a rated voltage of the shunt that can be different from the rated voltage of the connected bus. "step" simply scales active and reactive power linear with each additional step.

Results of both are tested against powerfactory, documentation is updated. If vn_kv is not specified in the create function, it defaults to the rated voltage of the bus.

The loss factor functionality is provided through the convenience function create_shunt_as_condensator as discussed in #25

closed 2017-05-04 by lthurner