e2nIEE/pandapower #23: nets equal error

FlorianShepherd, 2017-04-14:
toolbox.nets_equal results in an error if a compare a deepcopy version of net with net when numba = True/ False. Example:
```   
    import pandapower as pp
    import pandapower.networks as nw
    import pandapower.toolbox as tb
    net = nw.case4gs()
    net_copy = copy.deepcopy(net)
    pp.runpp(net_copy, numba=True)
    pp.runpp(net, numba=False)
    tb.nets_equal(net, net_copy, check_only_results=True, tol=1e-4)
```

Even though the results and dataframes seem to be equal if I compare manually

lthurner, 2017-04-14:
There is a check if the keys of the dataframes are the same. The non numba version has the array _closed_bb_switches, the numba version does not for some reason.

I am not really sure if that check really makes sense though. At least I feel it should be restricted to all keys that do not start with "_"

ascheidl, 2017-04-14:
For loadflows with r_switch != 0 the numba version of build_bus falls back on the non numba version (which uses _closed_bb_switches for the r_switch != 0 case).

I agree with @lthurner, we should exclude any internal variables and dataframes "_*" from the test.

FlorianShepherd, 2017-04-19:
Fixed by commit https://github.com/lthurner/pandapower/commit/a2ab021b6013d0da3cef6899d857527b4b5ed35c

closed 2017-04-19 by FlorianShepherd