e2nIEE/pandapower #24: pandapower 1.3

lthurner, 2017-04-19:
Already implemented for version 1.3:
- ZIP loads
- interactive plotting (plotly, mapbox)
- short circuit: kappa method C, full converter elements, speedups
- speedups in power flow

Further implementations needed before release:
- numpy 1.12 compatibility
- integrating pypower and making individual index files for pandapower
- convenience function for input of load, sgen, shunt
- shunt model?

jhmenke, 2017-04-25:
- Results for all elements (not just bus, line) are now available after state estimation
- Bugfix in bad data removal function for state estimation

Although the current implementation should be treated with caution as bad data will not be correctly . A caution box has been added to the documentation.

lthurner, 2017-05-02:
I have updated the feature list in more detail

@jhmenke 

@jkrstulo : are all issues with plotly/mapbox resolved or is there something you would like to do before the next release? Are the notebooks all finished?

@friederikemeier: when can you look into the OPF?

jkrstulo, 2017-05-02:
@lthurner : I believe all the issues are resolved. Some more user-testing would be beneficial, but for know I think it can be included in the new version. Only I would still like tomorrow to go through the tutorial to check if everything is clear and if all the plots show display as expected.

lthurner, 2017-05-03:
all tests are now passing with numpy 1.12, good job all around!

lthurner, 2017-05-10:
release is out, ZIP lods in OPF and pandapower 0.20 compatibility are left for the next release

closed 2017-05-10 by lthurner