e2nIEE/pandapower #26: PV generator

Tooblippe, 2017-04-25:
What would be the best way to model a fairly large 7MVA pv plant as a source

Sorry for posting here. I could not find the best place to ask questions.

Tobie

lthurner, 2017-04-25:
Do you mean as a current source? In that case it would be possible by creating a load and setting the constant current load to 100 percent in net.load.const_i_percent 

The ZIP load model is not yet included in the last pandapower release, so you have to use the current develop branch version for this to work.

lthurner, 2017-04-25:
If you are simply asking if PV plants are modelled as generators or static generators: they are normally modelled as static generators, since PV plants do not control voltage like big power plants do.

Tooblippe, 2017-04-25:
Thank you for the response. I have been struggling with this one. I have modelled it as a static generator for now. Basically, it feeds via a long 6.6kV line to the secondary of a Network transformer. There is also a load at the network transformer. I am experiencing over voltages at the PV (static) sending end. From there the question. I was sure static is the way to go, but I need voltage control.

This plant will be 7MW, and I am sure the inverters will be able to control the voltage at the sending end (6.6kV via a step-up transformer). Would this change the argument as this is a fairly large grid tied system?

This is a great initiative. I have been checking for the past couple of years and found nothing in python. this is really cool. I believe this project, because of the great documentation, and usability will become something bigger!!

Do you have a message board or discussion forum somewhere?

TObie

lthurner, 2017-04-25:
Thanks for your positive feedback! We do not have a forum yet, but if user numbers go up it might be a good idea.

To the question: using a PV generator means the PV plant would be able to feed in as much reactive power as is needed to always keep the voltage constant. Such a voltage control is not common for PV plants, even if they are the size of several MW. 

There are different ways that a converter based generators can feed in reactive power, mainly:
- with a constant power factor --> this is easy to model, just calculate Q as a function of P
- with some kind of droop function, e.g. a Q(V) curve --> This could only be modelled by implementing a controller that iteratively adapts the Q setpoint until a valid working point is found

Tooblippe, 2017-04-25:
Thank you. 

Keep up the good work.

closed 2017-04-25 by Tooblippe