e2nIEE/pandapower #29: GUI discussion

Tooblippe, 2017-05-05:
HI,

As stated before I am quite excited about this project. In my country (South-Africa) we have many, many small Municipalities. The network plannning skills and departments is mostly challenged. This leads to the appointment of consultants and therefore it is difficult to train inhouse people.

The software for loadflow is beastly expensive.  I have used PSSE and PowerFactory and although they are great products so expensive. We have a great local product called ReticMaster for Radial networks and PowaMaster for meshed networks but it also costs quite a bit of money

**I want to start a discussion on the following.**

- [ ] Do people think a simple drag and drop system as a seperate project could help 
- [ ] What platform would be the best
- [ ] Something like Kivy? This is native python and cross platform
- [ ] D3, had python bindings
- [ ] QT?
- [ ] Web, HTML, and JS based system - this will make a server based deployment easy, and in my case (South-Africa easier to roll out. We have so many different versions of Windows, installation rules etc at our clients)
- [ ] Any ideas of the appropriate frame work.
- [ ] Should the platform be Python based?

3. I found this example and it is what I have in mind. Its not free but will ive you an idea. I am searching for the appropriate OpenSource framework - http://freegroup.github.io/draw2d_js.app.brainbox/

4. Is there any interest in this group to get a proof of concept of the ground? I understand that the core of this project needs development so I do not mind starting a project, driving it and trying to get dev's to help?

regards Tobie

lthurner, 2017-05-05:
Hi Tobie,

you are right a pandapower GUI would be great! We have also talked about it quite a few times, but our main focus is scientific applications with high degree of automation. Since we use mainly scripting, there was just no time to get into developing a GUI. Your help would be greatly appreciated!

We already implemented some more specialized GUIs using PyQT. Using PyQT is very convenient, since it interfaces very well with the matplotlib collections. So all functionalities that are already available for regular matplotlib plotting (see [here](http://pandapower.readthedocs.io/en/v1.2.2/plotting.html)) can be easily adopted and made interactive in a widget. Maybe I can provide a simple example to show what I mean...

Furthermore, there is already a great project called [GridCal](https://github.com/SanPen/GridCal), which provides a PyQt GUI based on the pypower bus-branch model. This could at the very least be taken as an inspiration, maybe it even makes sense to take the GridCal code and try to adapt it internally so that it is based on pandapower instead of pypower?

About pypower: with the release of pandapower 1.3, the dependency for pypower in the power flow will be removed, since our development has diverged significantly from pypower. For now, we still have dependencies for the OPF, but that might also be removed in the future.

Best wishes
Leon

lthurner, 2017-05-05:
I have included an example consisting of a short script and and .ui file in the tutorial folder: https://github.com/lthurner/pandapower/blob/develop/tutorials/slider_example.py

This widget plots the MV oberrhein network and provides sliders to scale load and generation. The plot is automatically updated with the colormap of currents and voltages in the network. I think there is also a way to include pyqt widgets in jupyter, maybe I will look into that to make a nice example...

The .ui file is made with QT designer and defines the layout of the widget. As you can see, the plotting is done with the already available functions from the plotting module, since the matplotlib collections can be drawn directly in pyqt. In that way, quite nice GUIs can be defined with relatively little effort.

Best
Leon

P.S.: I know I didn't adress all the issues you have raised (e.g. platform, web based etc.), there are certainly a lot of things to consider. I wanted you to give an overview first about our thought process concerning a GUI so far.

Tooblippe, 2017-05-08:
HI,

Thank you for the feedback. I will have a look at it today. 

regards
Tobie

Tooblippe, 2017-05-08:
I have run the example and it looks good.  I dont know QT that well, but I assume Qt Designer (wich is a drag and drop app) was coded in Qt... thus it would e suitable to develop a drag and drop network modeler....

lthurner, 2017-05-08:
I have extended the slider example slightly so that it is now possible to click on buses and lines which opens a popup showing the name of the bus/line. Just to showcase how a matplotlib canvas can be made interactive. I have never used drag & drop with Qt but a quick google search suggests it is also possible...

Tooblippe, 2017-05-10:
I got some basics working.... I am just not a good enought coder to make this work quickly. But still doing some research

[image]

lthurner, 2017-05-10:
That looks pretty good for a start! Since this development is mostly seperated from the core development, I think it would be best to have a seperate repo for the GUI. Could you make a new repo and push your code there?

Tooblippe, 2017-05-10:
Will do....

Tooblippe, 2017-05-10:
What version of Qt should i start with... 4 or 5?

lthurner, 2017-05-10:
I have no strong preference one way or the other, but if there is no specific reason to use 4 I would suggest 5 since its the most recent version?

Tooblippe, 2017-05-10:
very basic 
https://github.com/Tooblippe/pandapower_gui

lthurner, 2017-05-12:
About the drag & drop system: I definitely think this is the way to go in the long run to make a nice and user friendly GUI. But to get a concept of the ground, it seems to be very complicated to implement connecting elements to buses, especially branches, which have a variable length.

Is it maybe easier to implement a simpler concept in the beginning, with a routine such as:

- right cklick on the canvas --> option to create bus
- right click on the bus --> options to create load, create sgen or create line
           - if a bus elemen is chosen, window pops up to input load parameter
           - if a branch element is chosen, wait for next click on a different bus to connect, then open window to input parameter

If this works, than maybe the step to a drag&drop system is also possible. Otherwise it seems that a lot of the initial effort goes into developing drag&drop?

Tooblippe, 2017-05-13:
HI, thank you. I am realising your comments as we speak...
I have made this in the mean time and it is proving helpfull

the load button for now just loads a demo file, then you can inspect and see html

it also embed an interpreter... it is becoming usefull

https://github.com/Tooblippe/pandapower_gui

Tooblippe, 2017-05-13:
[image]

[image]

[image]
[image]

lthurner, 2017-05-13:
Very nice! Good to see things are happening.

About the plotting: I created a function create_trafo_symbol_collection that creates transformer symbols with matplotlib collections (two circles, two lines). There is an example down below in the pandapower/plotting/collections.py, where a network is drawn with two lines and a transformer. As you can see, the only thing that is needed to draw a network like that are the bus coordinates.

So my idea would be this: all information that is needed to draw a net should be saved into the network (it needs to anyway, because a network also has to be drawn when loaded from the hard drive). When there is some kind of action (e.g. a new bus is created), the GUI writes the information into the net (e.g. creates a bus with coordinates) and then simply triggers a function that redraws the network from scratch. With the collections, this should be fast enough that it doesn not become slow even for larger networks. If the plotting is based on the collections, we can use a lot of functionility that is already implemented (as shown in the slider example with clicking and colormaps)

Tooblippe, 2017-05-13:
That is a good proposal. The format for this I think should come from the pandapower project as it controls the way networks are saved and loaded?  I  will focus on a while on the getting the basics of the app as it is better for now. For instance, I can push the 'net' variable to the interpreter, but changes in the interpreter for "net" is not sent back to the app.  the 'Build' tab is reserved for something like your proposal above. We can work out the details for that in time.

Tooblippe, 2017-05-14:
I have made some more changes. You can now add Loads and Busbars in the Network Builder tab. Still very basic but working. I am only using the minimum required parameters for now. Still want to add lines.
I have embedded the help in the build window as reference

I have made some more changes. You can now add Loads and Busbars in the Network Builder tab. Still very basic but working. I am only using the minimum required parameters for now. Still want to add lines.
I have embedded the help in the build window as reference

[image]

lthurner, 2017-05-14:
When I try to run the current version, I get an error:

Maybe we should open an issue on the new repo and continue the discussion there?

lthurner, 2017-05-14:
I have created a basic version of the network builder and created a [pull request](https://github.com/Tooblippe/pandapower_gui/pull/1). 

Maybe we should open an issue on the new repo and continue the discussion there?

Tooblippe, 2017-05-15:
@lthurner regarding this error - "**_Multiple incompatible subclass instances of InProcessInteractiveShell"_**. 

> Are you starting it up with `ipython `or `python.` Try starting it up with the `python pandappower_gui.py` command. I think `IPython `already embeds the interpreter, therefore, the error. I just ran it with `ipython `and got the same error. when I run it with `python `I have no issues. I have tested most on Windows and Ubuntu. Will check on my Mac later this week. I must say having the embedded interpreter is realy cool. You do the basics with the loading etc, but then move over to the interpreter. It is a full QT widget and you can plot in it as well. We can also start up a `Jupyter Notebook` and embed that.

**_Regarding the pull request and future discussions on this_**

> I am just travelling for a day and will then look at your pull request. I think we can move the discussion to the new repo.. Should I close this issue? And then we move over to the repo?

tobie

Tooblippe, 2017-05-15:
ps - what is an easy way to switch between the 1.3 and dev branches locally?

lthurner, 2017-05-15:
Ok discussion continues on the new repo: https://github.com/Tooblippe/pandapower_gui/issues 
But lets leave this issue open so people who are interested are pointed in the right direction

> ps - what is an easy way to switch between the 1.3 and dev branches locally?

The master branch is at pandapower 1.3, so you can switch between master / develop to switch between 1.3 and the most recent dev version

Tooblippe, 2017-05-15:
could you get it to run?

closed 2017-07-03 by lthurner

manuvarkey, 2022-11-21:
Hi,
I have written a GUI for pandapower ([https://github.com/manuvarkey/GElectrical](https://github.com/manuvarkey/GElectrical)) if any one is interested. The application is intended for MV/LV distribution networks but may be useful for other applications also. Currently application is in beta release phase.

Kalandoros, 2024-06-22:
I recently found a nice GUI for pandapower called Electrical Grid Simulator (EGS):
[https://github.com/aloytag/electrical-grid-simulator](https://github.com/aloytag/electrical-grid-simulator)

It looks very promissing and could be a good entry point for beginners. 
Since it is an educational institute a cooperation might be beneficial for both and would be a step further for this project.

kglikk, 2024-06-22:
There is also electrisim.com application which has pandapower(and OpenDss)
as backend and  GUI. Please check it out.