e2nIEE/pandapower #30: switches at three winding transformers

lthurner, 2017-05-05:
switches at three winding transformers are not yet implemented, but the create_switch function did accept "t3" as valid value for the element type. A switch with et="t3" would have no effect however even if open.

I adapted the create function so that an exception is raised ([here](https://github.com/lthurner/pandapower/blob/develop/pandapower/create.py#L1629)) as a quick fix. Of course in the long run we should implement switches for three winding transformers...

WinfriedL, 2018-05-08:
Is anybody working on this? Sooner or later I need switches at three winding transformers...  How difficult would it be to implement this?

ascheidl, 2018-05-08:
The handling of switches is one of the "darker" corners of the codebase and needs some overall refactoring. Since I have more time now, I want to look into this at some point in the near future.

For the meantime you could just add anotzher bus und use a bus-bus switch. (If necessary, you could give the switch a name that allows to identify the tranformer or add another column where you store the trafo_id). Is this a possible interim solution for you ?

ascheidl, 2018-07-24:
Interesting question.

I wrote a small test to compare both variants.

I created a random network with 1000 buses and 5000 lines.

Without bus-line-switches _pd2ppc(net) takes 6.3 ms. 
When I add 1000 open switches _pd2ppc(net) takes 29 ms. Adding another 1000 switches increases the time to 55ms.

However, if I create two closed bus-bus-switches at every line the time for _pd2ppc(net) is 10.8 ms (opening some of them should lower the time - so I didnt look into that)

So it seems your second option wins.

WinfriedL, 2018-07-24:
> I wrote a small test to compare both variants.

Thanks!

Inspired by your tests I did some testing on my own. In parts I can confirm your results.

I tried to test my worst case scenario, which would be that every line (or some other kind of branch) is connected at both ends by three switches and three auxiliary buses to the real busbars. For about 1% of the lines I added an open switch at one end.

For a testgrid with 1000 buses and 2000 lines I got 75 ms with the first option and 65 ms with the second option. Overall slower than your tests but my development machine is not very fast. But the second option is faster.

For a testgrid with 10.000 buses and 20.000 lines the first option needed 171 ms and the second option 642 ms. So for that case the first option seems to be faster. Here pandapower needs to merge 120.000 auxiliary buses and this is really the worst case. In reality only part of the branches is connected in this way (about 20%), many branches have only one or two switches at their ends.

It seems like from a performance perspective there is no clear winner for all cases, so it is still more a design choice if auxiliary buses should be in the pandapower network or not...

ascheidl, 2018-07-24:
OK, I created a similar network (10k buses, 20k lines, 120k switches) and indeed the time for _pd2ppc(net) is 100ms. (Is you PC 6 times slower than mine ?)

I had a deeper look because I was suprised that fusing buses should take so long. And it seems that fusing buses is not the main cause as it only takes less than 10% of the time.

Instead, the huge number of buses and switches increase the runtime at different parts of the ppc creation process. There is no immediate "fix". Overall refactoring and implementing hotspots in numba is the way to go.

WinfriedL, 2018-07-25:
>Is you PC 6 times slower than mine ?

That would not have surprised me, but in fact numba was set to false. With numba I get similar results, 130ms for 10k buses, 20k lines, 120k switches.
But in my tests most of the times the first option is faster, for example I  get only 70-90 ms for a 10k bus 20k line network with 1k open switches.

So my at the moment my preferences would be:
1. If in the foreseeable future switches at 3w transformers are supported, I tend to stick with my current decision to not have any auxiliary buses in the pandapower structure. (This would lead to ~15k buses, ~20k branches, ~1k open switches)

2. If that is not the case and I have to use auxiliary buses for 3w transformers, I would rather use the original auxiliary bus structure for all branch types. (This would lead to ~135k buses and 120k switches)

3. If I have to do that, I would at least think about creating the exact same structure for other elements as well (generators, loads, etc. are connected to auxiliary buses as well in my data) to have a consistent data portation. (This would lead to ~200k buses, ~230k switches)

>Instead, the huge number of buses and switches increase the runtime at different parts of the ppc creation process. 

As you can see from my numbers, every speedup for large numbers of buses is welcome...

That said, the current speed is completly acceptable for me at the moment. If I do a lot of calculations I hope to be able to reuse the ppc structure anyway... Not sure if this is currently working.

lthurner, 2018-10-12:
Switches at three-winding transformers are now implemented, a test/example is given here: https://github.com/e2nIEE/pandapower/blob/b5401ec41f703451da2a6f2de3985c0bb2857219/pandapower/test/loadflow/test_scenarios.py#L354-L407

I am closing this for now, if there are any problems with the implementation feel free to reopen

closed 2018-10-12 by lthurner