e2nIEE/pandapower #35: pandas 0.20

lthurner, 2017-05-09:
pandas released a new version and apparently it breaks the from_pickle functionality with an Import Error: https://travis-ci.org/lthurner/pandapower/jobs/230239640

I have downgraded to pandas 0.19 for the Travis tests for now. Lets see if we can fix this before the 1.3 release

omus, 2017-05-09:
For now could you make a patch release where the setup.py specifies "pandas>=0.17.0,<0.20.0"?

lthurner, 2017-05-09:
yes we will do that in the pandapower 1.3 release, which should go out tomorrow

Tooblippe, 2017-05-12:
I get this when I try and import Cigre cases or some others. Is this the same issue?

[image]

lthurner, 2017-05-12:
yes this is this issue, downgrading to pandas 0.19 should fix that

lthurner, 2017-05-20:
I have looked into this problem a little more: the issue is that we don't use the pandas function DataFrame.to_pickle but rather use the pickle library directly to save a dictionary that contains DataFrame objects. The pickle library saves the DataFrames as objects, and if that object definition changes, the unpickling fails.

A solution would be to convert all dataframes into dictionaries before saving them, and converting them back when loading. Will look into it now

lthurner, 2017-05-21:
I changed the to/from pickle functions so that only native python data types and no pandas objects are stored. Column sorting and dtypes are also saved to reconstruct the dataframe. Works now with pandas 0.20 and should also prevent future compatibility problems.

.p files that have been saved with a previous version of pandapower have to be resaved with the current version before udpating to pandas 0.20. This is a code snippet that loads and saves all networks in a certain folder:

```
for root, dirs, files in os.walk(my_dir):
    for file in files:
        if file.endswith(".p"):
            net = pp.from_pickle(os.path.join(root, file))
            pp.to_pickle(net, os.path.join(root, file))
```

The networks included in pandapower are already updated to the newest version and can be loaded with either pandas version.

Can anyone who experienced this issue please report if this solves the problem?

ascheidl, 2017-05-31:
The new implementation of to_pickle creates huge files. At the moment we have 260 MB in the networks folder.

A fresh clone of the repo has a size of 730 MB. I think this is way to much for some Python files and some documentation.

We should:

1. Try to find a more efficient way to save networks.
2. Clean up the git history from large files
2. If 1. fails maybe create a dedicated repo for networks

lthurner, 2017-05-31:
You are right, the files are larger then before, I didn't pay attention to that.

I created the dictionaries with the standard setting of to_dict, which saves it as {column -> {index -> value}}. That might be the reason, because every value gets its own sub dictionary.  There is also the possibility to store it as {column -> [values]}, maybe that will lead to smaller sizes

lthurner, 2017-05-31:
Changed the orientation to "split", which increases file sizes significantly in commit 70f06edd7a43dd34b1d3f3b36d044a6bafdae3df. The largest pegase network with over 9000 buses is down from 65MB to 17MB in filesize.

ascheidl, 2017-05-31:
The same file can be stored in 4.5 MB with to_excel.

There must be a better way.

What about my two other proposals?

lthurner, 2017-05-31:
I dont' know a way to make it more efficient without loosing pandas 0.19/0.20 compatibility, unless you have an idea?

If excel is more efficient we can also store it as excel in the networks folder, which would have the additional benefit that it is human readable for new users.

I think cleaning up git history and making a seperate repo for network files is a seperate issue. Cleaning up git history is ok by me, but the git repo size does not worry me to much, because most users don't clone the repo. But for pip install pandapower we should think about if it makes sense to download all networks? Maybe there is a way to make that optional?

lthurner, 2017-06-01:
An argument against using excel as the standard format is that it requires a package that is not part of the python distribution (xlrd)

joelmac, 2017-06-16:
I'm having a similar issue with pandapower 1.3.0.

I installed with pip3 install pandapower and get the following:

[image]

[image]

Not sure what other information might help or if there is an issue with my install.

lthurner, 2017-06-16:
Hi, this is not a similar issue, it is exactly this issue. You have three options:
1. downgrade your pandas installation to 0.19
2. instead installing pandapower through pip, use the most recent version from github (instructions see [here](http://pandapower.readthedocs.io/en/develop/getting_started/installation_develop.html))
3. wait until 1.3.1 is released - I actually wanted to it this week, but have been to busy...

lthurner, 2017-06-16:
pandapower 1.3.1 is released, so installing through pip should now also work with pandas 0.20. If anyone still experiences problems, please reopen.

Since the pickle files are not really efficient with regard to file size, we might switch to another standard format. Excel needs non-standard python libraries and saving also takes a long time. SQL databases might be better suited, see #46

closed 2017-06-16 by lthurner