e2nIEE/pandapower #37: html representation

lthurner, 2017-05-10:
I have made some changes to the html function contributed by @mdipierro in commit 191b2347e05e3150cd79ae6e037f0493332558cb:

- changed the name to to_html to be consisten with to_pickle, to_excel etc.
- changed the identation of adding the lines (they were never added for respect_switches=False)
- added a show_table parameter so showing the tables can be disabled --> i did this because for large networks the tables take up a lot of space and it takes a lot of scrolling to get to the graph. maybe an alternative would be to position the graph at the top?
- removed the l variable for impedances since it does not exist for impedances
- removed whitespace

@mdipierro can you look over those changes and comment?

mdipierro, 2017-05-11:
All looks good to me. Thanks. This can be improved a lot, also the part that does the graph should be taken out so that it can be embedded in a notebook. I may do so if I have time.

lthurner, 2017-08-14:
I will close this since there is currently no specific issue here, feel free to reopen when relevant