e2nIEE/pandapower #38: invalid documentation

hajef, 2017-05-11:
create_ext_grid has invalid doc for it's parameter va_degree
"**va_degree** (float, default 0.) - name of of the external grid*"
compare
"**name** (string, default None) - name of of the external grid"

This is the one mistake I came across until now but bugs seldon come alone and maby someone should take a closer look at the doc in general.

It is no big deal but I liked pp (and python in geral) for beeing well documented and it would be a shame if it could not get rid of this clay feed.

lthurner, 2017-05-11:
Yes I am confident this is a problem we can get rid of... In fact you can do that yourself: the table is located in /doc/elements/ext_grid_par.csv. If you correct the description there and commit your changes, it will be displayed correctly.

> This is the one mistake I came across until now but bugs seldon come alone and maby someone should take a closer look at the doc in general.

We already put quite some effort into the documentation, but you are very welcome to do have a look at it and propose improvements.

hajef, 2017-05-11:
I'd like to do so but I still get started with pp and don't know what should be the doc at that place. I just realized that two different parameters can't have the same doc and that a float can't contain a string.

commit ddf733e2 (2017-05-11): fixes #38
closed 2017-05-11 by lthurner