e2nIEE/pandapower #44: plotting raises deprecation warning

hajef, 2017-05-23:
Trying to plot a network with pandapower.plotting.simple_plot I get a deprication warning. See the message at file for details.
[DepricationWarning.txt](https://github.com/lthurner/pandapower/files/1021494/DepricationWarning.txt)
I tryed installing the mentioned igraph project but got the same warning.
I tryed to install jgraph and change "import igraph as ig" to "import jgraph as ig" but I got an AttributeError. Again details in file.
[AttributeError.txt](https://github.com/lthurner/pandapower/files/1021493/AttributeError.txt)

Comparing master and develop branche I learned that both use igraph but could not figure out how it is bound to the depricated one or which one is ment to be used here.

Is that my mistake, a problem of my IDE (brand new PyCharm) or a general one?

Tooblippe, 2017-05-23:
I got the same error just two hours ago. tried to build it into the GUI.

I used networkx but then asked me to install pyviz or something. then had to run out, and have not explored any further.

friederikemeier, 2017-05-23:
The package that you atually need is python-igraph.
[Python-igraph](https://pypi.python.org/pypi/python-igraph)
You can also install it via pip.

Tooblippe, 2017-05-23:
seems pip wont work on windows according to the documtation. on windows they point you to http://www.lfd.uci.edu/~gohlke/pythonlibs/#python-igraph for a whl

hajef, 2017-05-23:
Neither way work for me. Pip fails building whl and trying to install from wheel, it says that the wheel is not supported on this platform. I use Python 3.6 and the wheels are for 3.4 or 3.5 (tryed both) but shouldn't they be compatible? 
That's kind of a python-igraph problem but is there any other way to make pandapower.plotting work?

jhmenke, 2017-05-23:
Unfortunately igraph has to be compiled manually on Windows for Python 3.6 until it is available pre-compiled.

There is the option to use GraphViz to generate artificial geo coordinates instead of igraph: https://pandapower.readthedocs.io/en/develop/plotting/matplotlib/generic.html#pandapower.plotting.create_generic_coordinates

hajef, 2017-05-24:
I'm starting to think that my Anaconda ist broken. Can't install pygraphviz neither with pip nor as requirement in PyCharm. Looks like I have to parse my networks myself :/

jhmenke, 2017-05-24:
Why not install Python 3.5 parallel to your Python 3.6 until igraph works in 3.6?

hajef, 2017-05-24:
Won't work for me either. I can now install python-igraph using the wheel but I can't make it work all together.

jhmenke, 2017-05-24:
You also installed pycairo, right?

jhmenke, 2017-06-20:
Closed due to inactivity

closed 2017-06-20 by jhmenke