e2nIEE/pandapower #46: sql connection

lthurner, 2017-06-01:
Since the pandapower data structure is already close to a data base structure, it would make sense to provide a sql connection. It looks like the user changtse95 has already implemented something like this: 

https://github.com/lthurner/pandapower/compare/develop...chengts95:develop#diff-9c58788ccf37efa2fc19cbbc73a3a756

changtse95 can you comment on how mature this functionality is? Would you be willing to provide tests and contribute this to pandapower?

ascheidl, 2017-07-08:
I used changtse95 code as a basis to develop a first version for the sql connection. I tried to avoid having SQLAlchemy as dependency. However, keeping the dtypes of the dataframes correct after storing and reloading an networks didnt work out of the box (for example sqlite3 has not boolean type). So I create a DataFrame with all dtypes of all columns of all dataframes in the network and store it additionally in the database.

This might be necessary for other interfaces as well: to_json and to_excel tests fail when a network like networks.case115 is used.

Moreover, as a intermediate step the network is converted to a dictionary of dataframes. We should refactor the other methods (like to_excel) to use this dict of pds in order to avoid the root of all evil.

jhmenke, 2017-07-14:
original issue solved?

>This might be necessary for other interfaces as well: to_json and to_excel tests fail when a network like networks.case115 is used.

i implemented this in to_json/from_json

chengts95, 2017-07-16:
I used SQLAlchemy because it is  mature and universal, if you need help with the database IO, I am very glad to provide assistance.

lthurner, 2017-08-14:
Closed for now since there is no specific issue

closed 2017-08-14 by lthurner