e2nIEE/pandapower #47: ImportError: cannot import name 'runpp'

daniel-brettschneider, 2017-06-06:
Hi,

I have tried all installation methods from the documentation, but importing the pandapower module (`import pandapower`) always results in:

> ImportError: cannot import name 'runpp'

My current environment is

> python --version
> Python 3.6.1 :: Anaconda 4.4.0 (64-bit)
> pip freeze | grep pandapower
> pandapower==1.3.0

where pandapower is directly installed from the development branch.

lthurner, 2017-06-06:
Hi, 

we are currently preparing version 1.3.1, it is possible that there is still some problem with the current setup.py on develop. Try installing either:

1. The latest version with pip install pandapower
2. cloning the most recent version of the develop branch, but instead of using setup.py to install, add  the /pandapower folder to the python path

daniel-brettschneider, 2017-06-06:
I've already tried with and without setup.py and installing with pip for Python 2.7, 3.5, and 3.6. Can't get it running.

lthurner, 2017-06-06:
that is strange... Do you work with the Anaconda distribution? Can you please post the full error traceback?

daniel-brettschneider, 2017-06-06:
test.py
`import pandapower`

```
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    import pandapower
  File "/home/daniel/tools/pandapower/pandapower/__init__.py", line 5, in <module>
    from pandapower.diagnostic import *
  File "/home/daniel/tools/pandapower/pandapower/diagnostic.py", line 21, in <module>
    from pandapower.run import runpp
  File "/home/daniel/tools/pandapower/pandapower/run.py", line 10, in <module>
    from pandapower.optimal_powerflow import _optimal_powerflow
  File "/home/daniel/tools/pandapower/pandapower/optimal_powerflow.py", line 10, in <module>
    from pypower.add_userfcn import add_userfcn
  File "/home/daniel/Downloads/pypower.py", line 4, in <module>
    import pandapower.plotting
  File "/home/daniel/tools/pandapower/pandapower/plotting/__init__.py", line 6, in <module>
    from pandapower.plotting.plotly import *
  File "/home/daniel/tools/pandapower/pandapower/plotting/plotly/__init__.py", line 4, in <module>
    from pandapower.plotting.plotly.pf_res_plotly import *
  File "/home/daniel/tools/pandapower/pandapower/plotting/plotly/pf_res_plotly.py", line 13, in <module>
    from pandapower import runpp
ImportError: cannot import name 'runpp'
```

lthurner, 2017-06-06:
>   File "/home/daniel/Downloads/pypower.py", line 4, in <module>
>     import pandapower.plotting

This looks strange. pandapower is trying to import pypower (the module), but it looks like with your setup pypower is a file (pypower.py). The pypower module does not contain a file pypower.py. Check if you installed pypower correctly and maybe rename/move that file.

daniel-brettschneider, 2017-06-07:
Thanks, there was a broken install of pypower in an old python version. Now it's working.

closed 2017-06-07 by daniel-brettschneider