e2nIEE/pandapower #48: estimation bugs

jhmenke, 2017-06-09:
There are some issues with the estimation module which should be solved in the coming time:

- [ ] transformers are causing problems. The bus admittance matrices Ybus, Yf, Yt seem to be a little different for estimation compared to power flow. Debugging is required. 
- [ ] wls_matrix_ops should be refactored to use the ppc and its matrices more
- [ ] bad data removal (rn_max) is not reliable all the time
- [ ] line current measurements seem to prevent convergence. there is a whole chapter on these problems in the "Power System State Estimation" book.

I don't expect to work much on this this year though. Any help from potential users is highly appreciated.

jhmenke, 2017-07-17:
the Ybus matrices are the same, so that can be ruled out as the cause of non-convergence

lthurner, 2017-07-18:
Do we have tests of these cases of non-convergence and other cases that lead to problems? Maybe they could be included in the test suite but skipped for now as known bugs / problems to avoid the test suite failing entirely?

jhmenke, 2017-07-18:
The only bug was with the phase-shifting transformers, which is now fixed with commit dfa39edc19270bf8.

The rest are inherent limitations of the WLS SE, r_n_max test in their current implementations. Anyone is free to improve them.

jhmenke, 2018-06-05:
only res_bus_est, res_line_est, res_trafo_est, res_trafo3w_est are explicitly written now. the functionality will stay limited to this for the time being.

jhmenke, 2018-12-04:
there has not been any interest in optimizing the current estimation module. 
bad data and a line current test are implemented in the dev_2.0 branch, although the number of tests could be improved (as always).

i'm closing this issue due to lack of interest.

closed 2018-12-04 by jhmenke