e2nIEE/pandapower #50: Bug in Optimal DC Power Flow

ghamkhari, 2017-06-19:
The command `pandapower.rundcopp()` does not work and results in the following error:

`Traceback (most recent call last):`

  `File "<ipython-input-22-cc482b66fe4a>", line 1, in <module>
    runfile('/Users/Mahdi/Drive/Lab/Python/PandaPower_Tutorial/opf_basic.py', wdir='/Users/Mahdi/Drive/Lab/Python/PandaPower_Tutorial')`

  `File "/Volumes/Transcend/Applications/miniconda2/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile`  
 `execfile(filename, namespace)`

  `File "/Volumes/Transcend/Applications/miniconda2/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 94, in execfile`
    `builtins.execfile(filename, *where)`

  `File "/Users/Mahdi/Drive/Lab/Python/PandaPower_Tutorial/opf_basic.py", line 169, in <module>
    pp.rundcopp(net)`

  `File "/Users/Mahdi/Drive/Lab/Python/pandapower/pandapower/run.py", line 346, in rundcopp
    _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)`

  `File "/Users/Mahdi/Drive/Lab/Python/pandapower/pandapower/optimal_powerflow.py", line 48, in _optimal_powerflow`
    `result = opf(ppci, ppopt)`

  `File "/Users/Mahdi/Drive/Lab/Python/pandapower/pandapower/opf/opf.py", line 175, in opf
    results, success, raw = opf_execute(om, ppopt)`

  `File "/Users/Mahdi/Drive/Lab/Python/pandapower/pandapower/opf/opf_execute.py", line 63, in opf_execute`
    `results, success, raw = dcopf_solver(om, ppopt)`

  `File "/Users/Mahdi/Drive/Lab/Python/pandapower/pandapower/opf/dcopf_solver.py", line 226, in dcopf_solver`
    `opt['cplex_opt'] = cplex_options([], ppopt)`

  `File "/Users/Mahdi/Drive/Lab/Python/PYPOWER/pypower/cplex_options.py", line 94, in cplex_options`
    `lpmethod = ppopt['CPLEX_LPMETHOD']`

`KeyError: 'CPLEX_LPMETHOD'`

ghamkhari, 2017-06-19:
I note that, the command 
`pandapower.rundcopp(net, OPF_ALG_DC=200)  # using PIPS as the solver` 
 works, however the commands 
`pandapower.rundcopp(net, OPF_ALG_DC=500)  # using CPLEX as the solver`
`pandapower.rundcopp(net, OPF_ALG_DC=700)  # using Gurobi as the solver`
both fails. I am interested to run the last two commands (which fail), since the performances of `CPLEX` and  `Gurobi` are much better than that of `PIPS`.

lthurner, 2017-06-19:
We never tested pandapower with CPLEX or Gurobi, so its not surprising this doesn't work. However, since we do not implement our own OPF but simply provide an interface to pypower, it should be possible to make this work assuming it works in pypower. Do you know if the CPLEX and Gurobi work with pypower?

Regarding the specific problem: did you try to specify the missing parameters as in: pp.rundcopp(net, OPF_ALG_DC=500, CPLEX_LPMETHOD=2, ...)?

ghamkhari, 2017-06-19:
I haven't worked with `pypower` directly, since I couldn't find its documentation. If you have a link to the latest version of `pypower` and  its documentation, would you please share that with me? 

I just tried specifying the missing parameters: 

`pp.rundcopp(net, OPF_ALG_DC=700, GRB_OPT=0, GRB_METHOD=1, GRB_TIMELIMIT=1000, GRB_THREADS=1)`

Now, the `pandapower` part of the code works fine, but an error occurs in `pypower`: 

`
  File "/Users/Mahdi/Drive/Lab/Python/PYPOWER/pypower/qps_gurobi.py", line 240, in qps_gurobi
    gurobipy(c.T, 1, AA, bb, contypes, xmin, xmax, 'C', g_opt)`

`TypeError: 'module' object is not callable`

From the above lines, it seems that the file `qps_gurobi.py` in `pypower` is using a module `gurobipy()`, however `gurobipy` is not defined in as a function in `pypower`. Would you please let me know how can I report this bug to `pypower` developers?

ghamkhari, 2017-06-19:
I also note that, for reproducing the error in my previous comment, one has to first resolve some bugs in the `pandapower/pandapower/opf/dcopf_solver.py` file. If pandapower team are interested, I can `commit` the modified `dcopf_solver.py` to the `pandapower` github repository.

lthurner, 2017-06-19:
Yes please commit your changes as pull request, fixes are always appreciated!

pypower does not have an indepedent documentation as far as i know. Since it is a port of matpower, the matpower documentation is sufficient in most cases. However, not all parts of the code are tested and fully working. Some parts of the code might exist because they have been translated from matlab, but have never been used / tested in python. That might be the case here.

The pypower github repository can be found here: https://github.com/rwl/PYPOWER

lthurner, 2017-07-18:
Closed for inactivity, please feel free to reopen if there is something new on this issue

closed 2017-07-18 by lthurner