e2nIEE/pandapower #53: KeyError: max_p_kw

joelmac, 2017-06-26:
Hi,

I'm attempting to go through the tutorials on the website including the basic power flow tutorial.

I have just upgraded to pandapower 1.3.1 as a suggestion from a previous bug.

I have pandapower version 1.3.1 and pandas version 0.20.2.

The following is the error I'm receiving after following the steps in the `tutorial:`

`archimedes@archimedes-desktop:~/PandaConvNet$ python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pandapower
fr>>> from pandapower import networks as nn
>>> import pandas
>>> import pypower
>>> net = nn.example_simple()
>>> net.bus
                 name  vn_kv type  zone  in_service
0           HV Busbar  110.0    b  None        True
1         HV Busbar 2  110.0    b  None        True
2  HV Transformer Bus  110.0    n  None        True
3  MV Transformer Bus   20.0    n  None        True
4         MV Main Bus   20.0    b  None        True
5            MV Bus 1   20.0    b  None        True
6            MV Bus 2   20.0    b  None        True
>>> net.gen
        name  bus    p_kw  vm_pu  sn_kva  min_q_kvar  max_q_kvar  scaling  \
0  generator    5 -6000.0   1.03     NaN     -3000.0      3000.0      1.0

   in_service  type
0        True  None
>>> pandapower.runopp(net,verbose=True)
Traceback (most recent call last):
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandas/core/indexes/base.py", line 2442, in get_loc
    return self._engine.get_loc(key)
  File "pandas/_libs/index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc (pandas/_libs/index.c:5280)
  File "pandas/_libs/index.pyx", line 154, in pandas._libs.index.IndexEngine.get_loc (pandas/_libs/index.c:5126)
  File "pandas/_libs/hashtable_class_helper.pxi", line 1210, in pandas._libs.hashtable.PyObjectHashTable.get_item (pandas/_libs/hashtable.c:20523)
  File "pandas/_libs/hashtable_class_helper.pxi", line 1218, in pandas._libs.hashtable.PyObjectHashTable.get_item (pandas/_libs/hashtable.c:20477)
KeyError: 'max_p_kw'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandapower/run.py", line 292, in runopp
    _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandapower/optimal_powerflow.py", line 38, in _optimal_powerflow
    ppc, ppci = _pd2ppc(net)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandapower/pd2ppc.py", line 89, in _pd2ppc
    _build_gen_ppc(net, ppc)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandapower/build_gen.py", line 22, in _build_gen_ppc
    _build_gen_opf(net, ppc)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandapower/build_gen.py", line 355, in _build_gen_opf
    _copy_p_limits_to_ppc(net, ppc, eg_end, gen_end, _is_elements['gen'])
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandapower/build_gen.py", line 373, in _copy_p_limits_to_ppc
    ppc["gen"][eg_end:gen_end, PMIN] = -net["gen"]["max_p_kw"].values[gen_is_mask] * 1e-3 + delta
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandas/core/frame.py", line 1964, in __getitem__
    return self._getitem_column(key)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandas/core/frame.py", line 1971, in _getitem_column
    return self._get_item_cache(key)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandas/core/generic.py", line 1645, in _get_item_cache
    values = self._data.get(item)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandas/core/internals.py", line 3590, in get
    loc = self.items.get_loc(item)
  File "/home/archimedes/.local/lib/python3.5/site-packages/pandas/core/indexes/base.py", line 2444, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
[... 145 lines cut ...]
unity-scope-chromiumbookmarks 0.1
unity-scope-colourlovers      0.1
unity-scope-devhelp           0.1
unity-scope-firefoxbookmarks  0.1
unity-scope-gdrive            0.7
unity-scope-manpages          0.1
unity-scope-openclipart       0.1
unity-scope-texdoc            0.1
unity-scope-tomboy            0.1
unity-scope-virtualbox        0.1
unity-scope-yelp              0.1
unity-scope-zotero            0.1
urllib3                       1.21.1
usb-creator                   0.3.3
wadllib                       1.3.2
wcwidth                       0.1.7
webencodings                  0.5.1
Werkzeug                      0.12.2
wheel                         0.29.0
widgetsnbextension            2.0.0
xdiagnose                     3.8.5
xkit                          0.0.0
XlsxWriter                    0.9.6
zope.interface                4.3.2
`

friederikemeier, 2017-06-27:
Hi, what exactly should the OPF do in this case? Before running an OPF you need to specify some operational constraints and variables like the "max_p_kw" for generators. This example will work:
`import pandapower as pp
from pandapower import networks as nn
net = nn.example_simple()
net.gen["max_p_kw"] = net.gen.p_kw
net.gen["min_p_kw"] = net.gen.p_kw*2
net.load["controllable"] = False
net.sgen["controllable"] = False
pp.create_polynomial_cost(net, 0, "gen", np.array([[-1 ,0]]))
net.bus["max_vm_pu"] = 1.025
net.bus["min_vm_pu"] = .975
pp.runopp(net,verbose=True)
print(net.res_gen)
print(net.res_bus)`

You could also have a look at our OPF tutorial: https://github.com/lthurner/pandapower/blob/master/tutorials/opf_basic.ipynb
If you have further questions, feel free to ask!

Hope this helps. Kind regards!

friederikemeier, 2017-06-27:
Also, could you please let us know, which tutorial you were using? So we can improve it ;)

lthurner, 2017-06-27:
Maybe it would also be helpful to check for all constraints that have to be defined and raise an exception if they don't exist, e.g. "Please specify maximum active power for the generator in net.gen.max_p_kw" so that the user knows this is not a bug but that the value has to be specified.

joelmac, 2017-06-27:
https://github.com/lthurner/pandapower/blob/develop/tutorials/powerflow.ipynb

Now I see though the command was runpp, not runopp.    Thank you for helping me promptly though.  I very much appreciate it.

I second the suggestion by lthurner above.

friederikemeier, 2017-06-27:
Yes, that's a good idea. I will add the checks.

friederikemeier, 2017-06-30:
While adding the checks, I figured that max_p_kw and min_p_kw are by definition optional parameters, that should be set to +- p_lim_default when not specified. I would add a check to build_gen, if max_p_kw is in net.gen.columns and copy the  p_lim_default to the ppc in this case. If I do this, all tests still pass.
Your case OPF example here would still not converge, but the error would be a different one. 

What are the opinions? Should max_p_kw be optional?

lthurner, 2017-07-03:
sgen and load should not be controllable by default, because static generators and loads are not normally controllable. If the controllable column does not exist, it is assumed that all elements are static. Limits should have to be set for gens, sgens and loads, but not for ext_grids.

commit b9e64383 (2017-07-03): Added checks for necessary parameters in OPF
Refactored OPF documentation
updated docstrings in create.py
closes #53
closed 2017-07-03 by lthurner