e2nIEE/pandapower #58: numba dependency

lthurner, 2017-07-25:
As raised in this pull request #57 by kdheepak, pandapower can currently not be imported without numba. We should either

1. Find a way to continously test if pandapower works with numba (e.g. by uninstalling numba in one python version and checking if tests still pass)
2. Give up the idea that pandapower runs with and without numba and just make numba a mandatory dependency

The problem  with numba is that it is a C library that can't be easily installed through pip, which is why we wanted to keep it optional. On the other hand, it would be a lot of work and if people use anaconda, installing numba is not a problem.

Opinions?

ascheidl, 2017-07-25:
I strongly vote for 1.

FlorianShepherd, 2017-07-25:
I also vote for 1. Maybe with the exception of some features, like the connectivity check. But the PF and OPF should work without numba...

Is it possible to automate the test? Uninstalling and Installing numba all the time seems a little bit cumbersome

lthurner, 2017-07-25:
I have taken a practical approach and configured travis to uninstall numba in the python 3.6 test. There might be a more elegant way, but that works for now.

The tests are failing with an emport error as reported: https://travis-ci.org/lthurner/pandapower/jobs/257236441

I did my part in breaking the tests, now someone else please fix them :)

FlorianShepherd, 2017-07-25:
I tried to use the hack from "Sean Vieira": 
https://stackoverflow.com/questions/29587317/python-numba-jit-conditional-and-recursive-stack-use

But lots of tests are failing for different reasons if numba is not installed then...

jhmenke, 2017-07-26:
>The problem with numba is that it is a C library that can't be easily installed through pip

Although this is probably only true for Windows, this realistically only leaves option 1. Testing can be done in the (upcoming?) CI environment.

lthurner, 2017-07-26:
> There are many useful Python (Winpython etc.) environments, i don't think requiring Anaconda is the best way.

winpython also includes numba

lthurner, 2017-07-27:
numba is also availble on Christoph Gohlkes site as a wheel by the way: http://www.lfd.uci.edu/~gohlke/pythonlibs/#numba

So to summarize:
- everyone who uses anaconda or winpython has numba preinstalled
- everyone who uses some other distribution that does not include numba ca
       - install numba through pip if they are on linux
       - download numba from gohlke if they are on windows

So it seems to me we are keeping duplicate code so that people who are on windows using a distribution without numba and can't / don't want to install numba through gohlke can have a slower version of pandapower? That seems like overkill to me. I vote for 2.

FlorianShepherd, 2017-07-27:
Pandapower is now working without numba. 

But execution is slow, since numba is avoided by using a dummy JIT decorator. A warning message is given in that case and it is recommended to install numba. See commits:

25690283ed29f4c10579867717c595e16b68a0f7
4f912a772cdda69352463b247fab1d4edaa0a639
e57a07470b8ee9352949831796cb073e706f1a0b
391150b74385dc9aca350c6fdf915dff2b64692c
cd0f09deef0a5306b98d8ae9cf9ca3c4726bb122
04f67a118348a824d35583c25cf685f76956dafb

lthurner, 2017-07-27:
I would still like to discuss how we proceeed. There are two ways to get numba compatibility:

1. have functions that are optimized for numba (everything runs in loops), but are always used even if numba is not installed.
2. reimplement all numba functions as vectorized versions so that we have two different functions that do exactly the same, one optimized for numba, one optimized for numpy

1. is okay for me, since it is not a lot of additional effort. 2. means we would have to implement and maintain two functions that do exactly the same, and to me that does not seem worth it, since everybode can install numba if they really want (see above).

jhmenke, 2017-07-28:
Canopy, another Python distribution, also includes numba. 
If we required numba, this would definitely make it easier to maintain code and enable the optimization of other code as well. The downside is that there will be a pip exception if someone uses the regular Python and numba is added in the requirements. This might discourage new users.

lthurner, 2017-08-14:
yes we will for now use option 1. where numba functions are used without numba to make numba compatibility easier. Of course if anyone wants to reimplement numba functions in a vectorized way, they are more than welcome to do so

closed 2017-08-14 by lthurner