e2nIEE/pandapower #59: opf basic example fails

nesnoj, 2017-07-28:
I ran the first part of your [simple OPF example](https://github.com/lthurner/pandapower/blob/master/tutorials/opf_basic.ipynb) until `pp.runopp(net, verbose=True)` - it fails stating

```
Traceback (most recent call last):
  File "xxx/python/pypower/pandapower_test.py", line 11, in <module>
    pp.runopp(net=net)
  File "xxx/virtualenvs/misc/pypower/lib/python3.5/site-packages/pandapower/run.py", line 403, in runopp
    _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
  File "xxx/virtualenvs/misc/pypower/lib/python3.5/site-packages/pandapower/optimal_powerflow.py", line 38, in _optimal_powerflow
    ppc, ppci = _pd2ppc(net)
  File "xxx/virtualenvs/misc/pypower/lib/python3.5/site-packages/pandapower/pd2ppc.py", line 65, in _pd2ppc
    net["_is_elements"] = aux._select_is_elements_numba(net)
  File "xxx/virtualenvs/misc/pypower/lib/python3.5/site-packages/pandapower/auxiliary.py", line 354, in _select_is_elements_numba
    if net["_options"]["mode"] == "opf" and net._is_elements is not None:
  File "xxx/virtualenvs/misc/pypower/lib/python3.5/site-packages/pandapower/auxiliary.py", line 157, in __getattr__
    cls=self.__class__.__name__, name=key
AttributeError: 'pandapowerNet' instance has no attribute '_is_elements'
```
Any hints on this?

FlorianShepherd, 2017-07-28:
I just tried a clean install with 1.4.0 and the OPF test script and it seems to work for me. 

Have you created an empty network?

Is the result "True" if you try this code? 
```
import pandapower as pp
net = pp.create_empty_network()
print(net["_is_elements"] == None)
```

nesnoj, 2017-07-31:
Thank's for your reply!

Sorry, my bad! pyCharm was tricking me.. The example works out now!
Originally I wanted to do an OPF on the case2869pegase test case to get a better understanding on how fast pandas'/pypower's OPF is.

If I got it right the only things missing in the test network are:
1. The costs
2. Make loads static (not controllable)

So what I did to run an OPF on the case2869pegase test network:
```
import pandapower as pp
import pandapower.networks as pn
import numpy as np
import pandas as pd

net = pn.case2869pegase()

# costs
costeg = pp.create_polynomial_cost(net, 0, 'ext_grid', np.array([1, 0]))
for id in range(0,509):
    costgen = pp.create_polynomial_cost(net, id, 'gen', np.array([1,0]))

# make loads not controllable
net.load.loc[:, "controllable"] = pd.Series()
net.load.loc[:, "controllable"] = False

pp.runopp(net, verbose=True)
```
This results in the exception above..
I probably missed something out here?!

I'd be very grateful for hints..

friederikemeier, 2017-07-31:
Hi!
You also need to specify, whether the sgens are controllable:

`# make sgens not controllable
net.sgen.loc[:, "controllable"] = pd.Series()
net.sgen.loc[:, "controllable"] = False
`

nesnoj, 2017-07-31:
Thanks!
But I still got the same exception
```
...
AttributeError: 'pandapowerNet' instance has no attribute '_is_elements'
```
It might be something more fundamental? In the [docs](https://pandapower.readthedocs.io/en/v1.4.0/powerflow/ac.html?highlight=_is_elements) `_is_elements` seem to be related to time series calculation..

FlorianShepherd, 2017-08-01:
You're right, there was a bug in line 354 in pandapower/auxilary.py when a net has not _is_elements, for example if you load it from a pickle file...
See: b68c976cfa2274bc632d35110a82bf48596103dd 

You can pull the current develop branch to get the fix.

nesnoj, 2017-08-01:
Yupp, works out now! (although it does not converge)
Thanks for tracking down this one! :+1:

closed 2017-08-01 by nesnoj

lthurner, 2017-08-11:
For the sake of completeness: the reason it does not converge seems to be the voltage constraint, which is set to a maximum of 1.1 pu in the pegase network. It converges with a maximum voltage of 1.2 pu