e2nIEE/pandapower #60: issue loading case9241pegase

nesnoj, 2017-08-01:
The pickle of case `case9241pegase` seems to be broken in `develop` branch as
```
import pandapower.networks as pn
net = pn.case9241pegase()
```
throws
```
Traceback (most recent call last):
  File "xxx/python/pypower/pandapower_case2869pegase_opf_test.py", line 10, in <module>
    net = pn.case9241pegase()
  File "xxx/git-repos/pandapower/pandapower/networks/power_system_test_cases.py", line 596, in case9241pegase
    case9241pegase = pp.from_pickle(os.path.join(_get_cases_path(), "case9241pegase.p"))
  File "xxx/git-repos/pandapower/pandapower/file_io.py", line 187, in from_pickle
    net = read(f)
  File "xxx/git-repos/pandapower/pandapower/file_io.py", line 178, in read
    return pickle.load(f, encoding='latin1')
_pickle.UnpicklingError: could not find MARK
```

FlorianShepherd, 2017-08-03:
In my setup (Windows 7 / Ubuntu Linux 64bit, Python 3.6, Anaconda 4.3) it works.

"could not find MARK" sound like a unpickling error. I checked my version with "pickle.format_version" and it is 4.0

Maybe this is the problem? Which version are you using?

nesnoj, 2017-08-03:
On a fresh pp install (v1.4.0) in a virtualenv it works fine! But using the manual install of the cloned pp-repo with pip3 (using "-e" Option) I get the described exception in the `master` as well as in the `develop` branch. Since the versions of all installed packages equal (including pickle v4.0), the issue might be related to recent modifications of pp?!
I run Python 3.5.3 on Xubuntu 17.04.

ascheidl, 2017-08-03:
This might indeed be a reason.

However, as soon as I find some time I will alter the storage format of the included networks from pickle to json or sqlite anyways.

These formats are faster and smaller.

nesnoj, 2017-08-03:
Ok.
Update: The `case9241pegase.p` files of both versions are different (says MD5)..
I will simply use the file from the current stable..

closed 2017-08-03 by nesnoj