e2nIEE/pandapower #61: OPF: simulation of network including batteries over time snapshots

nesnoj, 2017-08-02:
Dear developers,

I'd like to add a battery and do simulations over several time snapshots.
In your code I only found some batteries in the [CIGRE example as non-controllable static generators](https://github.com/lthurner/pandapower/blob/f2aecd7d09e93bb95954ac93b7278b9dc09467be/pandapower/networks/cigre_networks.py#L255). I added a battery using a controllable generator.
Is it possible to simulate a battery in pp over several time steps - does pp actually support OPF over time?

I couldn't find any SOC-related code but to be sure: I assume the answer is no?

FlorianShepherd, 2017-08-03:
Unfortunately scripts and functions for the simulation of time-dependent variables, such as batteries, are not integrated in the pandapower open-source version. PP is designed for single power flow, opf, short circuit and state estimation analysis. As you already mentioned, batteries could be implemented with a controllable sgen.  Feel free to open a branch / project, which can serve as a basis for such implementations and to get the help of others facing similiar problems.

Nonetheless, pandapower is well suitable for time series simulations. Parts of it are even optimized for time series simulations. For example the "recycle" parameter can be used to reduce the amount of (unncessary) conversion steps. Also the different init method reduce the time to converge for the NR power flow algorithm to speed up simulations.

nesnoj, 2017-08-03:
Thanks for your statement, I dug around some of the functionalities..
Currently I do not have time for forking but I'll consider that for future works.
Nevertheless, thanks for your great tool which enriches the open source modelling world!

closed 2017-08-03 by nesnoj