e2nIEE/pandapower #66: 11 tests fail with Linux

jhmenke, 2017-09-04:
Eleven tests are failing with pandapower 1.4.0 and pandas 0.19. Many more are failing with pandas > 0.19
I have tried both Python 2.7.10 and 3.6.

> ../pandapower/test/api/test_auxiliary.py .
> ../pandapower/test/api/test_create.py FF
> ../pandapower/test/api/test_diagnostic.py .....................
> ../pandapower/test/api/test_file_io.py ...
> ../pandapower/test/api/test_std_types.py ..........
> ../pandapower/test/api/test_toolbox.py .....
> ../pandapower/test/converter/test_from_mpc.py .
> ../pandapower/test/converter/test_from_ppc.py ....
> ../pandapower/test/converter/test_to_ppc_and_mpc.py .
> ../pandapower/test/estimation/test_wls_estimation.py .............
> ../pandapower/test/loadflow/test_results.py ..................
> ../pandapower/test/loadflow/test_rundcpp.py .F..
> ../pandapower/test/loadflow/test_runpp.py .F..................
> ../pandapower/test/loadflow/test_scenarios.py .......
> ../pandapower/test/networks/test_cigre_networks.py ...
> ../pandapower/test/networks/test_create_example.py ..
> ../pandapower/test/networks/test_kerber_extreme_networks.py ..........
> ../pandapower/test/networks/test_kerber_networks.py ..........
> ../pandapower/test/networks/test_mv_oberrhein.py .
> ../pandapower/test/networks/test_power_system_test_cases.py ..................
> ../pandapower/test/networks/test_simple_pandapower_test_networks.py ....
> ../pandapower/test/networks/test_synthetic_voltage_control_lv_networks.py FFFFF
> ../pandapower/test/opf/test_basic.py .........F.
> ../pandapower/test/opf/test_costs_mixed.py .
> ../pandapower/test/opf/test_costs_pol.py ..
> ../pandapower/test/opf/test_costs_pwl.py .....
> ../pandapower/test/opf/test_costs_pwl_q.py ....
> ../pandapower/test/opf/test_curtailment.py .
> ../pandapower/test/opf/test_dcline.py ...
> ../pandapower/test/opf/test_oberrhein.py .
> ../pandapower/test/opf/test_opf_cigre.py .
> ../pandapower/test/shortcircuit/test_gen.py ...
> ../pandapower/test/shortcircuit/test_impedance.py ..
> ../pandapower/test/shortcircuit/test_meshing_detection.py ....
> ../pandapower/test/shortcircuit/test_motor.py .
> ../pandapower/test/shortcircuit/test_ring.py ...
> ../pandapower/test/shortcircuit/test_sgen.py ...
> ../pandapower/test/shortcircuit/test_trafo3w.py ..
> ../pandapower/test/shortcircuit/test_transformer.py .....
> ../pandapower/test/topology/test_graph_searches.py .F
> 
> == 11 failed, 206 passed in 298.59 seconds ==

lthurner, 2017-09-04:
tests are also failing on travis: https://travis-ci.org/lthurner/pandapower

Apparently there is an unexcepted import of pplog, @rbolgaryn is looking into it

jhmenke, 2017-09-04:
I'm refering to 1.4.0. from July 2017. I think Roman's commit might have introduced additional issues.

lthurner, 2017-09-04:
If so it has nothing to do with Linux. Travis tests on Linux: https://travis-ci.org/lthurner/pandapower/jobs/270809300#L1232

jhmenke, 2017-09-04:
Okay then it seems to be specific to the machine.

jhmenke, 2017-09-04:
Were the pandapower generic networks not updated in 1.4.0 for pandas 0.20? i still get pandas index errors on a different machine.
installed via pip.

closed 2017-09-04 by jhmenke