e2nIEE/pandapower #68: Keep J after power flow

lthurner, 2017-09-11:
J is sometimes needed after a power flow for voltage stability studies. @FlorianShepherd  put together a tutorial on how it can be retrieved: https://github.com/lthurner/pandapower/blob/develop/tutorials/advanced/get_Ybus_and_J.ipynb
That is however quite some effort to reconstruct something that was already constructed and discarded just a few lines of code before. Is there any reason against giving an option to keep the last J from the powerflow?

friederikemeier, 2017-09-19:
With my next commit you can acces J after a newton pf (lgorithm = "nr" , default) like this:

```
    pp.runpp(net)
    J_intern = net._ppc["internal"]["J"]
```

closed 2017-09-19 by lthurner

lthurner, 2017-09-19:
I also added a tutorial on how to access ppc, Ybus and J and how to use the mappings for bus indices:

commit 61e91ad0 (2017-09-19): Made J accessible after Newton Raphson Powerflow. closes #68
commit df8e461b (2017-09-19): Made J accessible after Newton Raphson Powerflow. closes #68