e2nIEE/pandapower #72: Can't import pandapower

mihagibson, 2017-10-02:
Hi!
I have problems importing pandapower. Whenever I run import pandapower it starts running the code but nothing happens until I cancel it.

I have Anaconda and Windows 10.

Please help me with your solution.

FlorianShepherd, 2017-10-02:
Hi mihagibson,
have you tried the tests? 
Try: 
```
from pandapower.test import run_all_tests
run_all_tests()
```

Which editor do you use?
Best regards
Florian

mihagibson, 2017-10-03:
First, thank you for your reply.

I use Jupyter, but it also doesn't work if I run it in PyCharm. I tried importing the test as you suggested, but it's the same. It starts running and then the star [*] just won't go away until I interrupt the kernel.

I hope that we will manage to find the solution since this PandaPower looks really awesome.

ascheidl, 2017-10-03:
Hi,

first you should try with a fresh Anaconda environment if you didnt.

If it is still not working you could edit the pandapower/__init__.py and comment out all lines. The import should work now, because no code is executed at import time. Then you can remove the comments one by one to identify the "bad" import statement in the __init__ file. In this way you should be able to dig deeper to find the root cause (which is probably the import of a third party library).

Cheers,
Alex

mihagibson, 2017-10-03:
I tried a lot of stuff but didn't work. Then I reinstalled anaconda and now it works. But then I tried to do all the tests and apparently, opf doesn't work. My report is attached.

If everything else is gonna work properly, then it is ok, since I don't need opf calculations.

Thanks a lot for helping. 
Please, can you at least confirm me that only opf has problems?

[pp_report.txt](https://github.com/lthurner/pandapower/files/1353612/pp_report.txt)

lthurner, 2017-10-03:
All tests are passing (see 260 passed at the end), so everything is working fine, including the OPF. The error messages in between are loggings, not failing tests. The logging should however be disabled during testing, I am going to look into why those message pop up there....

closed 2017-10-03 by lthurner