e2nIEE/pandapower #73: flexible units: kW/MW/GW

lthurner, 2017-10-09:
Since pandapower was developed for distribution systems, all units are given in kW. Since it is increasingly used for High/Extra High voltage levels, we should think about if there is a way to allow the definition of power values with MW or GW?

mihagibson, 2017-10-13:
It would be really cool if you could set values manually.

lthurner, 2017-10-19:
The general problem is that units are hard coded into the column names. We decided to do it like that, so that units are directly obvious when writing code. I think that is one of the reasons pandapower is so convenient to use, and we should definitely keep it like that. But if we want to make units flexible, the column names would change, for example it could either be net.load.p_kw or net.load.p_mw. Someone writing a script accessing net.load.p_kw would not be able to run that script on a network working with MW. 

My suggestion is the following:
- introducing a parameter net.power_unit that signals if powers are in kW, MW, GW
- changing the power value in the create functions from "p_kw" to "p" and adding a second variable power_unit. If net.power_unit is different from the power_unit given to the create function, it throws an error. 
- Providing a conversion functions that automatically converts the power unit. So if someone receives a network in MW but wants to run a script written for kW on it, they can use this function to convert the network and then run their script

This would obviously be a big change to the API, so we should really think it through and discuss it.

ascheidl, 2017-11-13:
OK, I try to recap our discussion:

kW should stay the default, so nothing changes for the unaware user. Using alternative power units is for convinience of users who know what they to. So it is not a "... big change to the API ...".

We introduce net.power_unit as suggested. Depending on this parameter, pandapower expects the corresponding columns (p_kw, p_mw or p_gw and the corresponding unit for q)

For the api we identified serveral possibilities:
1. we just copy all functions - so we have 3 functions for create_load and automatically convert the values with respect to net.power_unit
2. we alter the function definitions depending on power_unit (we monkeypatch all functions)
3. we move the p_* into the kwargs
4. leons suggestion to alter "p_kw" to "p" 

As far as I remeber 3 and 4 we not really prefered.

Did I miss something ?

lthurner, 2018-11-26:
We have discussed this again in depths today and have come to the conclusion that allowing flexible units is just to much effort and would open up the risk of incompatible developments, which we don't want to risk.

However, we have discussed changing the base unit to MW, since that is the unit that is best readable and produces the least zeros over all voltage levels:

> p_kw = 10 #LV load
p_kw = 400 #LV/MV Transformer
p_kw = 40000 #HV/MV Transformer
p_kw = 600000 #EHV/HV Transformer

> p_mw = 0.01 #LV load
p_mw = 0.4 #LV/MV Transformer
p_mw = 40 #HV/MV Transformer
p_mw = 600 #EHV/HV Transformer

If we are already making a non-backwards compatible release with the aim of making pandapower better suited for transmission systems, this might be a sensible change.

WinfriedL, 2018-12-13:
Just tried out the new units. While I am in favour of the new units, are you sure that you want to change typicaly small parameters as well? For example pfe_kw/pfe_mw for transformers is usually in a range more suitable for kw. On the other hand it might still be better to have the same units everywhere.

lthurner, 2018-12-13:
> For example pfe_kw/pfe_mw for transformers is usually in a range more suitable for kw. On the other hand it might still be better to have the same units everywhere.

Yes I also thought about this, but pfe_kw was the only parameter that I came across where this is the case and it didn't seem worth it to me to introduce an exception to a general rule for just one parameter. 

Staying with pfe_kw would also be fine with me, but if there is no other parameter with this issue i vote for uniformity.

WinfriedL, 2018-12-13:
I don't know if this a general rule, but the active power demand of shunt elements I have would also be better defined in kw than in mw. But all of this is not terrible in mw. I just want to mention it.

lthurner, 2018-12-17:
We have decided to stay with pfe_kw since it is more intuitive and the unit is also directly in the variable name, so there should be no issue with accidental mixups. For the shunt power we however want to stay consistent within one table (p_mw/q_mvar).

closed 2019-02-19 by lthurner