e2nIEE/pandapower #75: simple_plotly doesn't plot network according to matrices

mihagibson, 2017-10-11:
Hi!
I use Jupyter notebooks.
The code below is from the website and it is supposed to plot the MV network. But when I run the code below the network that is plotted is not the same as the one that I wanted to plot (from the website).
And also the network plotted with simple_plotly() does not correspond to the bus, line and trafos matrices. So the plotted network is not plotted as it is supposed to be. Please see the attached picture.

[image]

net_mv = pn.create_cigre_network_mv(with_der=False)
simple_plotly(net_mv)

jhmenke, 2017-10-11:
The plot is correct, i suspect that only the labels are set incorrectly.

jhmenke, 2017-10-11:
This was just a small bug with incorrectly ordered bus_geodata. It will be fixed in commit f316742fc15f6162fc.

mihagibson, 2017-10-11:
Thank u for your answer, but I am not used to github. So is it possible to download the whole cigre_networks.py and synthetic_voltage_control_lv_networks.py scripts?

or do I have to change them by myself?

lthurner, 2017-10-11:
You can install the current version from github by uninstall pandapower:

> pip uninstall pandapower

And then installing the newest develop version of pandapower from github:

> pip install git+https://github.com/lthurner/pandapower.git@develop

It will take a while to install because the repo is cloned into, but in the end you should have the newest version.

mihagibson, 2017-10-11:
I uninstalled it, then when I try to install it I get this:

Error [WinError 2] The system cannot find the file specified while executing command git clone -q https://github.com/lthurner/pandapower.git C:\Users\Miha\AppData\Local\Temp\pip-7s8t_zo7-build
Cannot find command 'git'

lthurner, 2017-10-11:
you have to install git: https://git-scm.com/

mihagibson, 2017-10-11:
I 've installed, buses are now OK but lines are still wrong.
Switch table also doesn't correspond to the plot.

jhmenke, 2017-10-12:
I checked it again and you are correct. Maybe the author @jkrstulo can identify the problem the fastest?

FlorianShepherd, 2018-05-28:
The error is now fixed I assume. See https://github.com/lthurner/pandapower/commit/f14386ceccd21ae24a2e51962767aced7a8fd80e 

 @mihagibson would you confirm that, please?

Also note that the legend in simple_plotly is a little bit strange atm. We'll have a look into that soon.

commit 3c6182e9 (2017-10-12): :bug:fix simple_plotly uses correct bus_geodata index (closes #75)
closed 2018-07-09 by lthurner