e2nIEE/pandapower #76: switches are not plotted

mihagibson, 2017-10-13:
Hi Guys!

I tried to plot the network from the tutorial called _create_simple.ipynb_.
When I try to plot the network plot doesn't show the switches.

It doesn't work for both simple_plot & simple_plotly.

[image]

lthurner, 2017-10-15:
Yes bus-bus switches are not plotted yet. This would be a good beginners task, at least for matplotlib collections. I suggest looking at the transformer collection function: https://github.com/lthurner/pandapower/blob/develop/pandapower/plotting/collections.py#L206 and deriving a similar function that plots connections between two buses that are connected with a closed bus-bus switch.

lthurner, 2018-04-23:
No this issue refers to bus-bus switches, the create_line_switch_collection plots bus-line switches. There is no create_collection function for bus-bus- switches yet, but it would be really simple to implement...

jhmenke, 2018-04-26:
I also implemented this in simple_plot, therefore i'm closing the issue now. The styling the helper line between two buses connected by bus-bus switches can be modified by the user (also made invisible).

[image]

closed 2018-04-26 by jhmenke

miek770, 2018-12-08:
Sorry for commenting on a closed issue but since it's not implemented in plotly yet, shouldn't this issue remain open? Otherwise I would suggest opening a new issue for plotly, because the interactive plot is much better from the static plot, apart from this missing feature.

P.S.: Tested on a github installation (from develop).

jhmenke, 2018-12-10:
plotly is not really supported as well as the other modules right now, because it was originally an external contribution. you could open an issue with the missing features of our plotly-plotting and maybe someone will take care of it. or maybe you could look into this if you have the time 😄

lthurner, 2018-12-10:
> Otherwise I would suggest opening a new issue for plotly, because the interactive plot is much better from the static plot, apart from this missing feature.

The matplotlib plots can also be made into interactive plots with PyQt. Here is an example for this (although I am not sure if it still works, haven't checked for some time): https://github.com/e2nIEE/pandapower/blob/develop/tutorials/slider_example.py

The matplotlib plots also have the advantage of being much faster than plotly, which can take quite some time to build for larger networks.

bakinloye, 2021-02-28:
Hello, is there a way to open and close switches?

rbolgaryn, 2021-03-02:
What do you mean by a way to close and open switches?