e2nIEE/pandapower #78: Include IEC 60909-4 network for short-circuit calculation

rwl, 2017-10-18:
Part 4 of IEC 60909 provides "Examples for the calculation of short-circuit currents". It would be interesting if pandapower included these among the existing collection of example networks:

https://pandapower.readthedocs.io/en/v1.4.1/networks.html

They could be used in the test suite to ensure compliance with the standard. The Conformity Assessment Scheme provides test configurations based on this model (MiniGrid) in CIM RDF/XML format:

https://www.entsoe.eu/Documents/CIM_documents/Grid_Model_CIM/CGMES_v2.4.15_TestConfigurations_v4.0.3.zip

lthurner, 2017-10-18:
I tested all the short circuit functionality against commercial tools (mostly powerfactory, but also Sincal), which I assume comply with the standard, because that was more convenient and flexible to handle. I agree that having the examples from the standard would be nice, but I currently don't have the time to do it. Of course if you or anyone else what be able to do it, that would be much appreciated.

Since the example grids are quite small, it might be easier to manually create them through the pandapower API than to parse the CIM file provided by ENTSO-E.

rwl, 2017-10-20:
I have added a test case for the IEC 60909-4, section 6 network here:

https://github.com/rwl/pandapower/commit/334ee18a217e5d5d412bf0d4bb0d161065190507

The results do not match up yet, but most of the input values should be correct.

FlorianShepherd, 2019-08-06:
Please reopen if there is any update on this @rwl

wangzhenassd, 2021-01-13:
@rbolgaryn do you think this example would be a good starting point for the generator-close shortcircuit calculation?

PR #1137 (2021-05-26): Multiple functional improvements in shortcircuit calculation
1. Added support generator-close sc calculation as well as on generation buses
2. Added support of power station calculation for OLTC and non-OLTC transformer (needs to be checked)
3. Removed single_sc calculation, since it is now integrated in calc_sc
4. Updated the internal program flow of calc_sc
2. Initialized the calculation of 1ph shortcircuit calculation with trafo3w in network
3. Added test case according to IEC 60909-4, which should close #78
4. Changed the convention of rdss_pu to rdss_ohm to generator, since in IEC 60909 r [... cut]
closed 2021-05-26 by lthurner