e2nIEE/pandapower #80: An error in plotly_maps.ipynb

WuXiaorui, 2017-11-05:
I tried to run plotly_maps.ipynb
it raised an error!
> ---------------------------------------------------------------------------
> NameError                                 Traceback (most recent call last)
> <ipython-input-3-8cd4aeba5d0e> in <module>()
> ----> 1 simple_plotly(net, on_map=True)
> 
> c:\users\sami\appdata\local\programs\python\python35\lib\site-packages\pandapower\plotting\plotly\simple_plotly.py in simple_plotly(net, respect_switches, use_line_geodata, on_map, projection, map_style, figsize, aspectratio, line_width, bus_size, ext_grid_size, bus_color, line_color, trafo_color, ext_grid_color)
>     103     line_trace = create_line_trace(net, net.line.index, respect_switches=respect_switches,
>     104                                    color=line_color, width=line_width,
> --> 105                                    use_line_geodata=use_line_geodata)
>     106 
>     107     # ----- Trafos ------
> 
> c:\users\sami\appdata\local\programs\python\python35\lib\site-packages\pandapower\plotting\plotly\traces.py in create_line_trace(net, lines, use_line_geodata, respect_switches, width, color, infofunc, trace_name, legendgroup, cmap, cbar_title, show_colorbar, cmap_vals, cmin, cmax)
>     296         line_trace = dict(type='scatter',
>     297                           text=[], hoverinfo='text', mode='lines', name=trace_name,
> --> 298                           line=Line(width=width, color=color))
>     299 
>     300 
> 
> NameError: name 'Line' is not defined

recounter7, 2017-11-08:
Same Error for me:

File "C:\anaconda\Lib\site-packages\pandapower\plotting\plotly\traces.py", line 297, in create_line_trace
    line=Line(width=width, color=color))

NameError: name 'Line' is not defined

lthurner, 2017-11-08:
I am guessing you don't have plotly installed. Try:
> pip install plotly

closed 2017-11-12 by lthurner