e2nIEE/pandapower #81: Error in plotting.simple_plot()

basemidlbi, 2017-11-17:
Hi Leon,

the simple plotting in pandapower (without coordinates) does not work.
I think the module igraph does not work, and you need to use jgraph instead.
Using the module networkx give an error as well. I wrote these simple commands to test simple_plot():

import pandapower.networks as ppn
kerber = ppn.create_kerber_dorfnetz(p_load_in_kw=6.)
import pandapower.plotting as ppl
ppl.simple_plot(kerber, library="igraph")

lthurner, 2017-11-17:
The commands as you put them plots the network like this for me:
[image]
So it does work with the right installation.

You need to install python-igraph (not igraph), which also needs pycairo.  Compiled wheels for both packages can be found on Christoph Gohlkes site: https://www.lfd.uci.edu/~gohlke/pythonlibs/

basemidlbi, 2017-11-17:
Hi Leon,
thanks for the fast reply. It works well after installing pycairo.
Best regards, Basem

lthurner, 2017-11-17:
You are welcome Basem!

closed 2017-11-17 by lthurner