e2nIEE/pandapower #83: pandas 0.21 compatibility

lthurner, 2017-12-04:
pandas has made some API changes in [pandas 0.21](https://pandas.pydata.org/pandas-docs/stable/whatsnew.html#whatsnew-0210-deprecations) which causes tests to fail. Seems like this breaks pandapower in several instances:

- parameter sheetname was renamed to sheet_name in read_excel
- apparently summing over an empty pandas series now returns nan instead of 0, which causes problems in consistency checks

lthurner, 2017-12-04:
- excel problem is solved
- consistency check is fixed
- int/float problem in merge_nets is fixed
- python 2/3 compatibility for graph_characterestics function in graph_search was fixed (unrelated to pandas 0.21)
- bug in get_connected_buses fixed (unrelated to pandas 0.21)

julffers, 2017-12-04:
- failing diagnostic_tests due to changed behaviour of boolean columns in pandas are fixed

lthurner, 2017-12-04:
All tests pass now, full compatibility with Python 2/3, networkx 1.x/2.x and pandas 0.19/0.20/0.21 is ensured

Time for pandapower 1.4.2!

closed 2017-12-04 by lthurner