e2nIEE/pandapower #84: error running Load FLow

mihagibson, 2017-12-07:
Hi!
I have a problem when running one of my scripts (Load Flow calculation). It worked on my PC and laptop until recently. But from yesterday I get the following error when running it on my PC, whereas it works on my laptop.

The error is below. I would be glad if you could help me.

IndexError                                Traceback (most recent call last)
<ipython-input-4-b57fc2c01b10> in <module>()
    153 
    154         # calculate state with PV
--> 155         pp.runpp(mv, calculate_voltage_angles=False)
    156 
    157 

C:\Anaconda3\lib\site-packages\pandapower\run.py in runpp(net, algorithm, calculate_voltage_angles, init, max_iteration, tolerance_kva, trafo_model, trafo_loading, enforce_q_lims, numba, recycle, check_connectivity, r_switch, voltage_depend_loads, delta_q, **kwargs)
    244     # net.__internal_options.update(overrule_options)
    245     net._options.update(overrule_options)
--> 246     _powerflow(net, **kwargs)
    247 
    248 

C:\Anaconda3\lib\site-packages\pandapower\powerflow.py in _powerflow(net, **kwargs)
     82         net["converged"] = True
     83 
---> 84     _extract_results(net, result)
     85     _clean_up(net)
     86 

C:\Anaconda3\lib\site-packages\pandapower\results.py in _extract_results(net, ppc)
     20 
     21     _get_bus_v_results(net, ppc)
---> 22     bus_pq = _get_p_q_results(net, bus_lookup_aranged)
     23     _get_shunt_results(net, ppc, bus_lookup_aranged, bus_pq)
     24     _get_branch_results(net, ppc, bus_lookup_aranged, bus_pq)

C:\Anaconda3\lib\site-packages\pandapower\results_bus.py in _get_p_q_results(net, bus_lookup_aranged)
    160                 bus_lookup = net["_pd2ppc_lookups"]["bus"]
    161                 bus_idx = bus_lookup[ppi]
--> 162                 i_cc = (ppc["bus"][bus_idx][:, PCID] + 1j * ppc["bus"][bus_idx][:, QCID])[l['bus'].values]
    163 
    164                 sl = - v * conj(i_cc) * net._ppc['baseMVA'] * 1e3

IndexError: index 4011 is out of bounds for axis 0 with size 3272

lthurner, 2017-12-07:
That looks like a ZIP load issue, are you using loads with a constant current part?
What version are you using, did you update to pandapower 1.4.2 or are you working with a develop version from github?

mihagibson, 2017-12-07:
I use loads with const_z_percent = 100.0 .
I had 1.4.1 version on both of my computers, but as I said it works only on laptop.

If I set const_z_percent to zero, then it works.

I upgraded to 1.4.2 version

If I type conda list in cmd, it says that I have pandapower 1.4.2, but if I use pp.__version__ in jupyter notebook it says that I use version 1.4.1

lthurner, 2017-12-07:
That sounds like you have two conflicting versions of pandapower in your python path. Check your installation and make sure you have only one instance of pandapower installed, and that it is the newest one (pandapower 1.4.2). Report back if the problem still occurs then

mihagibson, 2017-12-07:
Thank you for your immediate help! I had two different versions installed.
Anyway, this package is awesome!!!

lthurner, 2017-12-07:
Thank you, glad to hear it!

By the way, if you are modeling 100% impedance loads you can also use a shunt element instead of a load. It models an impedance and also gives you additional parameters like rated voltage and a step parameter.

closed 2017-12-07 by lthurner