e2nIEE/pandapower #85: parallel lines in multigraph

lthurner, 2017-12-18:
edges in multigraph for parallel lines currently are translated like this:
{
0: {"weight": 1, "key": 2, "element": "l"},
1: {"weight": 1, "key": 3, "element": "l"}
}

for two pandapower lines with indices 2/3 and length 1.

Wouldn't it be cleaner to implement it like this:

{
2: {"weight": 1, "element": "l"},
3: {"weight": 1, "element": "l"}
}

Calling the parameter "key" is also a bit unfortunate, because "key" is also the name of the networkx edge attribute, which is confusing and also leads to problems with networkx 2.0

jhmenke, 2018-01-29:
i wouldn't call it "idx" though. "line_index" is cleaner.

WinfriedL, 2018-07-12:
> Wouldn't it be cleaner to implement it like this:
> 
> {
> 2: {"weight": 1, "element": "l", "idx": 2},
> 3: {"weight": 1, "element": "l", "idx": 3}
> }

Cleaner, yes, but dangerous. While not common, it is possible that two branches of different type (e.g. line and impedance) connect the same two buses. Those can have the same index. So one of the networkx edge dictionary entries, where the key is a tuple (from_bus, to_bus, index), gets overwritten.
The chances might be slim, but not impossible.

> i wouldn't call it "idx" though. "line_index" is cleaner.

Not all edges are lines, and having different attribute names for different edge types seems like a bad idea to me.

Maybe just "index"?

lthurner, 2018-07-12:
> Cleaner, yes, but dangerous. While not common, it is possible that two branches of different type (e.g. line and impedance) connect the same two buses. Those can have the same index. So one of the [...]
The chances might be slim, but not impossible.

That is true, I had not thought about this case. But is there really a real life use case where a line and an impedance would be connected in parallel? Since impedances are usually used as equivalent elements, I would think this case does not really ocurr. But of course it is a theoretical possibility.

I would suggest checking for this case when adding impedances and throwing an error if the same edge already exists. This shouldn't have a large impact on performance, assuming impedances are used a lot less than lines?

> Not all edges are lines, and having different attribute names for different edge types seems like a bad idea to me.

> Maybe just "index"?

Yes I agree it should be the same for all edges. "index" is fine with me.

WinfriedL, 2018-07-12:
> But is there really a real life use case where a line and an impedance would be connected in parallel?

From my experience, if it is possible to create in a network model, somewhere it is. But there might be other cases as well. A 3W transformer where two windings are connected to the same bus would have the same key. Not real life, but with thousands of them in a network model, which has evolved over years, I would not exclude the possibility. Impedances and transformers could be parallel as well.

> assuming impedances are used a lot less than lines?

a lot less, yes. But in my case roughly 10000 lines vs 2500 impedances. 
In my case one would have to check 2500 impedances, 4500 transformers and 1500 three winding transformers with 3 edges each so again 4500 checks. Even if the check is fast, I don't see the advantage. More complex and possibly slower code, and cases which won't work (even if they might not be real life). And if the index is an edge attribute it can be easily accessed.

jhmenke, 2018-07-12:
> Not all edges are lines, and having different attribute names for different edge types seems like a bad idea to me.
>Maybe just "index"?

"index" might be used by the graph internally. 
"element_index" on the other hand would make it clear that this edge belongs to a pandapower element index.

lthurner, 2018-07-12:
> In my case one would have to check 2500 impedances, 4500 transformers and 1500 three winding transformers with 3 edges each so again 4500 checks. 

I wasn't thinking about a check on the pandapower data structure. Doing a check for "does this edge already exist in the networkx graph" just when adding impedance edges would cover all cases of imepdances parallel to lines, impedances parallel to three-winding transformers and impedances parallel to two-winding transformers. Since the networkx structure is basically a dictionary, this would come down to checking if a key is in a dictionary once for each impedance in the grid. Even for 2500 impedances (which I assume is an extreme use case) that should not impact performance a lot.

> Not real life, but with thousands of them in a network model, which has evolved over years, I would not exclude the possibility. 

That might be true, but not every case has to be supported. The user is also responsible for having a correct data model. For example, I wouldn't consider having a three-winding transformer connected to the same bus at both low voltage sides a case that needs to be supported.

> Even if the check is fast, I don't see the advantage. 

The advantage would be that the attributes can be directly accessed. Right now, if i want to identify the edge that belongs to line 3 in this structure:

{
0: {"weight": 1, "key": 2, "element": "l"},
1: {"weight": 1, "key": 3, "element": "l"}
} 

I have to iterate over all edges between those buses until I find the right one. With this structure:

{
2: {"weight": 1, "element": "l", "idx": 2},
3: {"weight": 1, "element": "l", "idx": 3}
}

I can directly access the right edge through the dictionary.

> "index" might be used by the graph internally.
"element_index" on the other hand would make it clear that this edge belongs to a pandapower element index.

I don't think "index" is part of the networkx nomenclature, "key" is used for identifiers

WinfriedL, 2018-07-12:
>just when adding impedance edges would cover all cases of imepdances parallel to lines, impedances parallel to three-winding transformers and impedances parallel to two-winding transformers.

I was thinking about the same, but for this the order of edge creation have to change. In the current implementation impedences gets created before transformers. If impedances are created last, you are right.

>I wouldn't consider having a three-winding transformer connected to the same bus at both low voltage sides a case that needs to be supported.

Not sure if I agree with that. But right now it is supported. By changing the index, it would not be supported any more.

>I can directly access the right edge through the dictionary.

I see the point. I was more thinking about the other direction, which would be easy. I don't know the common use cases for the graph, so I have no idea how often one would need that functionality.

BTW, I don't think any of this would be a problem for my cases right now. I hvae different indices for all element types anyway, so the first problem cannot occur for me. Only the 3W transformer problem might exist in the data. On the first look I don't have any of those.
So while I still would warn against it, if you decide having the index as the edge key I can live with it ;-)

> The user is also responsible for having a correct data model.

Not sure if I should comment this ;-)
Dealing with large data sets from different sources I would hate to have to modify them to make it work, especially if I know it could be working if the implementation would be a little different.

lthurner, 2018-07-16:
> I see the point. I was more thinking about the other direction, which would be easy. I don't know the common use cases for the graph, so I have no idea how often one would need that functionality.

The use case I had when I opened this issue was that I created a graph for a network, and then slightly changed the network (added a line, removed a line, opened a switch etc.) to optimize the grid. Instead of always creating a completely new networkx graph for the slightly changed grid, I wanted to directly update the graph with theses changes. That proved to be difficult since edges can't be easily accessed. 

>  Dealing with large data sets from different sources I would hate to have to modify them to make it work, especially if I know it could be working if the implementation would be a little different.

I know what you mean, it probably just comes down to the question of what is a valid use case. If having an impedance parallel to a transformer is a valid use case, then it might happen in larger data sets, even if its unlikely. But if it is not a valid use case, then it doesn't matter how large the data set is, this case should simply not occurr. And if it does ocurr due to a conversion error or faulty data, then we don't need to support it. Of course wether it is a valid use case or not is up for discussion, but until anyone can name me an instance where this actually makes sense, I remain sceptical.

I understand the implementation should be safe and it would be preferrable to have an implementation that works in absolutely all cases. But on the other hand, I don't think it is a good idea to have an implementation that makes our lives difficult in our actual use cases just to cover for some corner cases that might theoretically exist, but no one can really come up with a realistic example where it would.

lthurner, 2018-07-16:
Or maybe we can still come up with a solution that satisfies both requirements? I just checked and strings also work as edge keys. So how about:

{
"l2": {"weight": 1, "element": "l", "index": 2},
"l3": {"weight": 1, "element": "l", "index": 3}
}
?

WinfriedL, 2018-07-16:
I like the idea, that would solve most "problems". Since it is a dictionary key, a tuple should also work, like
{
("line",2): {"weight": 1, "element": "l", "index": 2},
("line",3): {"weight": 1, "element": "l", "index": 3}
}
In this way the element table and index is given and can be directly accessed.

This would not solve the 3W transformer problem. I still consider transformers where both mv and lv sides are connected to the same bus as a valid.

While I don't have that exactly in my models, I have several transformers connecting two identical low voltage busbars to a high voltage busbar. I see no reason why the low voltage sides could not be coupled by a switch, leading to a construction which is already very near the problematic case. Assuming some porting between models they could end up as a 3W transformer with two sides at the same bus.
It is theoretical, but I think it is valid...

Having something like that could help... but makes access to 3w trafo edges a bit more complicated:
{
("trafo3w_hv",1): {"weight": 1, "element": "t3", "index": 1},
("trafo3w_mv",1): {"weight": 1, "element": "t3", "index": 1},
("trafo3w_lv",1): {"weight": 1, "element": "t3", "index": 1},
}

lthurner, 2018-07-16:
We have talked about this again in quite some detail in our developers meeting today and have come to two conclusions:

1. We agree with the tuple solution that you have proposed, this is clean and eliminates the impedance problem.

2. With regards to the three-winding transformer, we have thought about in which case this would actually lead to a problem. Because even when there is a 3W-transformer with the same MV and LV bus, the connections between the buses are still correctly represented (the HV bus is connected to MV/LV bus), so that connectivity searches would not be impacted. The transformer also has no length, so that shortest-path-search operations or similar searches would also not be affected. The only case where this would actually have an impact would be graph-flow operations that rely on the new R/Z values with three-winding transformers that have the same MV/LV bus. And even if we would allow this case, it is not really certain that the impedances would be correct, because the MV/LV edge would be generated as a edge from and to the same bus with the MV/LV impedance of the transformer, and who knows what that would do with graph-flow algorithms. So in conclusion we don't believe it is worth to for this corner case to build a whole nomenclature around the trafo3w edges. So we want to stick with ("trafo3w", index) as a tuple, and if a user really encounters the case mentioned above they would have to manually update the attribute of the 3W-transformers. Unless we have missed a use case where that could lead to problems that we did not think of?

WinfriedL, 2018-07-16:
You are perfectly right. It does not matter for zero weights. Worst case would be a wrong impedance weight since one of the two paths would get overwritten in graph creation. Of cource one could also think about calculating a resulting impedance of the parallel connection, but at the moment I don't think it is worth the trouble.

> Unless we have missed a use case where that could lead to problems that we did not think of?

I can't think of any more problematic use cases. Thanks for discussing everything!

I can update the pull request #127 with the new keys. If you do not want to merge that until more tests are done (which I would understand), someone else might have to change the existing implementation accordingly...

lthurner, 2018-07-16:
> Thanks for discussing everything!

Thanks for providing the input! I think now we have arrived at a really good solution, shows its always worth discussing the details :)

> I can update the pull request #127 with the new keys. If you do not want to merge that until more tests are done (which I would understand), someone else might have to change the existing implementation accordingly...

If the code is going to be changed anyways I would preferr changing the new code and not the old one. So if you could add them to #127 I would try to set up some tests for the current implementation and then merge the new version if there are no problems with these tests.

By the way we have also discussed to keep the "element" and "index" variables out of the attributes dictionary entirely, because they are already in the dictionary key so that there is no duplicate information. So the end result should look somewhat like this:

{
("line",2): {"weight": 1, "R": 0.2, "X": 1},
("line",3): {"weight": 1, "R": 0.1, "X": 0.6}
}

julffers, 2018-12-11:
Can we include this in pandapower 2.0? #209

lthurner, 2019-02-19:
Implemented and merged in #127 as discussed above

closed 2019-02-19 by lthurner