e2nIEE/pandapower #87: External grid bus seems switched to internal bus between calculations leading to wrong results.

calleg, 2018-01-17:
We created a 0.4 kV grid containing several household nodes connected through lines coming from one distribution substation (0.4kV <->  20kV). The substation is modelled using a transformer (create_transformer) and an external 20kV grid (create_ext_grid). Something like this:
(Ext Grid)----(Bus 1)----(Transformer)----(Bus 2)----<further low voltage grid lines and nodes> 
Then, we use the calculations, as shown below. First we try a default calculation, before we try another calculation with specific parameters, if the calculation fails. However, somehow the bus 1 seems to become switched with bus 2, if the second calculation is approached. Meaning, the calculation is successful but wrong. To be correct vm_pu of bus 1 should be 1 and vm_pu of bus2 should be something else. This is switched, if the second calculation is executed. We already create a completely new net object to avoid some side effects, but it still happens.
```
try:
    net = build_pp_network(grid_data)
    pp.runpp(net)
except:
    try:
        net = build_pp_network(grid_data)
        pp.runpp(net, algorithm='nr', init='auto', max_iteration=100, tolerance_kva=1e-03)
    except:
	# error
```
Our python environment is deployed at an Ubuntu 16.04.3 LTS virtual machine with Python 3.5.2 -x86_64 having the following Panda Power related python packages installed:
•	decorator (4.2.1)
•	llvmlite (0.21.0)
•	matplotlib (2.1.1)
•	networkx (2.0)
•	numba (0.36.2)
•	numpy (1.13.3)
•	pandapower (1.4.2)
•	pandas (0.22.0)
•	pip (9.0.1)
•	PYPOWER (5.1.3)
•	scipy (1.0.0)
•	setuptools (38.4.0)
It seems to me that it is an issue with numba, but I am not sure. Any ideas, please?

lthurner, 2018-01-17:
Without a working minimal example that reproduces the problem, it will be difficult to guess what is going wrong. Two tips that might help you:

1. You say you are recreating the net entirely from scratch to make sure there is no dependece between the two nets. But it seems like you are using the same variable grid_data (whatever that is) in both functions. If the first call of build_pp_network changes something in grid data, the net might be different the second time. Try calling it with build_pp_network(copy.deepcopy(grid_data) instead) to see if it changes anything
2. There is a diagnostic function that checks the grid for consistency: http://pandapower.readthedocs.io/en/v1.4.2/powerflow/diagnostic.html Try running pp.diagnostic(net) might help you to find out whats wrong with the grid

calleg, 2018-01-19:
Thank you for your reply. I will try to create a code example, that I can replicate here.
Alternatively, I can offer a web session, where I can show you more, if you are interested.

lthurner, 2018-01-26:
I am going to close this for now, since the problem is not reproducable with the provided description. Please reopen if you have a working minimal example.

closed 2018-01-26 by lthurner