e2nIEE/pandapower #88: OPF don't converge

Alex-89, 2018-02-06:
Hello, 

I have a Neplan Grid and convert this throuh a access database to Python.

The most functions from pandapower are working well, except the optimal power flow (OPF).

The OPF don't converge if I have CHP in my grid. I have initialize the CHP as like follow:

```
pp.create_gen(net=net, bus=get_element_index(net, "bus", obj_TOPOLOGY.NODE1),
                     p_kw=-1*obj_CurrentObject.SR * obj_CurrentObject.COS*M_to_k,
                     vm_pu=1,
                     sn_kva=-1*obj_CurrentObject.SR*M_to_k,
                     name=obj_TOPOLOGY.NAME,
#                   index=index,
#                   scaling=,
                     type=obj_CurrentObject.TYPE,
                     controllable=True, 
#                   vn_kv=,
#                   xdss=,
#                   rdss=,
                     cos_phi=obj_CurrentObject.COS,
                     in_service=bool(obj_TOPOLOGY.SW1),
                     max_q_kvar=-1*sin(acos(obj_CurrentObject.COS))*obj_CurrentObject.SR*M_to_k,
                     min_q_kvar=0,
                     max_p_kw=-1*obj_CurrentObject.COS*obj_CurrentObject.SR*M_to_k,
                     min_p_kw=0)
```
When I use only loads as flexibility then its work without a problem. 

The error messages that i get:
> PYPOWER Version 5.0.0, 29-May-2015 -- AC Optimal Power Flow
> Python Interior Point Solver - PIPS, Version 1.0, 07-Feb-2011
> Numerically failed.
> Did not converge in 10 iterations.
> Traceback (most recent call last):
> 
> File "<ipython-input-229-779ec2121d57>", line 1, in <module> VPP_Network.calc_opf()
> 
> File "L:\10_Forschungsgruppen\04_FG_BKS\01_Projekte\10_WSW VPP (Virtuelle Kraftwerke.NRW)\04_ThemenCluster\Tools-Python\SQuOp\SQuOp\SQuOp_Network\SQuOp_Network.py", line 767, in calc_opf
>     pp.runopp(self.__Network, verbose=True, init=initialisation)
> 
>   File "C:\Users\ahobert\AppData\Local\Continuum\anaconda3\envs\VPP\lib\site-packages\pandapower\run.py", line 388, in runopp
>     _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
> 
>   File "C:\Users\ahobert\AppData\Local\Continuum\anaconda3\envs\VPP\lib\site-packages\pandapower\optimal_powerflow.py", line 58, in _optimal_powerflow
>     raise OPFNotConverged("Optimal Power Flow did not converge!")
> 
> OPFNotConverged: Optimal Power Flow did not converge!

Thanks for help!

friederikemeier, 2018-02-07:
Hi, this is because there may not be a solution to your OPF problem. Can you tell me which cost functions you used?
Does it converge before you add the sgen?
I see you have reactive power limits. Try to broaden them and check, to which setpoint of the reactive power the OPF converges.

Alex-89, 2018-02-08:
Hello,

I integrated following cost function for the gen:

`pp.create_polynomial_cost(net, ID, "gen", array([0.15,0]))`

I didn't know excactly the solution about the network ist too much.

You said sgen, to declare the CHP as gen is correct, isn't it? In the docu of pandapower is descirpe that controllable generation ist initilaize as gen.

To broaden the reactive power limits haven't any effect. 

Should I change the cost function?

friederikemeier, 2018-02-08:
Hi,

So, that looks ok. Please try setting the active power constraints broader? Does the OPF converge without the generator? Are there other controllables?
Is the network within the limits (min/max_vm_pu, max_loading for lines and trafos) when you run a simple runpp() ? 
Is max_p_kw and max_q_kvar positive?

Alex-89, 2018-02-13:
Hi, 
the active power constraints are from 100% to 0%. It oughta not any effect about the OPF.

I have equiped the grid with one controllable generator and with five controllable loads. 

I have a timebase powerflow with a controll function that detected all limit violation and "try catch" all not not converge oft the simple runpp(). 

By a simulation of one day I have some cases there not convert the runpp() and some cases ther I have a simple limit violation and in both cases the OPF don't converge.

Currently is max_p_kW = 0 and  max_q_kvar = Q, but I tried alleready max_p_kW = -P, it haven't any impact. It must be correct max_p_kW = 0 and min_p_kW = -P , isn't it? (It stand in the tutorial: http://nbviewer.jupyter.org/github/lthurner/pandapower/blob/develop/tutorials/opf_basic.ipynb)

friederikemeier, 2018-02-13:
Hi,

So without seeing the exact cases I can't help you. It should converge though if you have opened up all your constraints. If you could reproduce this in an example and post it here, we can help. 
Without the problematic network and your exact code I cannot explain the problem.

Alex-89, 2018-02-14:
Hi, 
the active power constraints are from 100% to 0%. It oughta not any effect about the OPF.

I have equiped the grid with one controllable generator and with five controllable loads. 

I have a timebased powerflow with a controll function that detected all limit violation and "try catch" all not not converge oft the simple runpp(). 

By a simulation of one day I have some cases there not convert the runpp() and some cases ther I have a simple limit violation and in both cases the OPF don't converge.

Currently is max_p_kW = 0 and  max_q_kvar = Q, but I tried alleready max_p_kW = -P, it haven't any impact. It must be correct max_p_kW = 0 and min_p_kW = -P , isn't it? (It stand in the tutorial: http://nbviewer.jupyter.org/github/lthurner/pandapower/blob/develop/tutorials/opf_basic.ipynb)

Alex-89, 2018-02-14:
Hello,

I have following cost function defined:

Alex-89, 2018-03-02:
Hello,

I have found my mistake.

Thanks for your help. I'll close this thread.

closed 2018-03-02 by Alex-89