e2nIEE/pandapower #93: Large indices lead to high memory consumption

WinfriedL, 2018-03-26:
Creating a network with large indices e.g. when creating generators leads to high memory consumption.

The problem is the line `lookup = -np.ones(max(pandapower_index) + 1, dtype=int)` when building the lookup:

https://github.com/lthurner/pandapower/blob/aef97f1f16ab496dd7d0b772c486b6e3be566b93/pandapower/pd2ppc.py#L255-L269

If one index is very large this creates a huge (mostly empty) ndarray, filling the memory. While I have not checked with the rest of the code this seemes unnecessary.

The same problem exists when building the bus lookup:
https://github.com/lthurner/pandapower/blob/aef97f1f16ab496dd7d0b772c486b6e3be566b93/pandapower/build_bus.py#L109-L117

For buses a workaround is to call `pandapower.create_continuous_bus_index(net, start=0)` before running the powerflow.

The problem only becomes significant for large indices of greater than 100000000 which might be unusual but occured in some of my calculations. For smaller indices it still seems like a waste of memory though.

ascheidl, 2018-03-26:
Our tests showed that the current implementation of the bus lookup (+numba) is the most performant one. So we decided to trade a slightly higher memory usage with increased performance (the resulting memory footprint is negligible for reasonable maximal bus indizes). 

In order to let unaware user know about this peculiarity we log a warning in case large bus indize values are encountered.

I was not aware that we use a similar implementation for element lookup. Maybe we should log a warning here too.

The problem should be easily solved by net.gen.reset_index(inplace=True) ?

WinfriedL, 2018-03-26:
Thanks for the explanation.
 If you have done the tests I am the last person who stands in the way of a performance increase by wasting a bit of memory, which in most cases should indeed be insignificant. However a warning would be nice.

But I wonder if you have considered very sparse indices in your performance tests. I would suspect that the performance/number of elements would decrease with a more sparse indices?

The data I am dealing with at the moment uses one continuus index for all types of network elements, which leads to a sparse index for one type of element.

Initally I used `create_continuous_bus_index` and ignored the index for all other elements to make it run at all, but this makes comparison with the original data very hard.

Luckily I can process the original data for a more compact index, since already for that data an index of > 100000000 is not necessary.

So I still would recommend a warning and I still might hit this problem in the future when dealing with larger data sets...

FlorianShepherd, 2018-04-03:
FYI, a little bit of history: The first implementation was with dicts and very low memory consumption, with 
key = pandapower index, val = ppc index

This was really slow for large arrays since it was necessary to iterate over every key in the dict. Since every index is unique we switched to the array where 
array_index = pandapower index
array_val = ppc_index

As ascheidl said, this is the fastest solution we found, but has problems with large indices.

Maybe a sparse lookup would be a compromise here. Thank you for the hint!

lthurner, 2018-04-03:
One additional note on using very high index numbers: a use case where this happens if you have some identifying number that you want to use as an index. I don't know if that's your usecase or not, but if so, an alternative to using the id as an index is to just use continuous indices when creating the net, and put the IDs in a new column, e.g. net.bus.ID. This way you avoid the memory problem, and you also avoid losing leading or trailing zeros. Because if you have an ID such as 0125810, this would become index 12581 and you might have trouble identifying it later on. If you save it in a new column as a string, all information can be preserved and it is easier to identify the ID later on.

lthurner, 2018-04-03:
Regarding the idea of a sparse lookup array, I tested it with a pandas SparseArray. The access times are significantly higher with the sparse solution, especially if not one value is accessed, but many at one time:

>dense = np.random.randn(1000000)
sparse = pd.SparseArray(dense)

>%timeit dense[500:1000]
631 ns ± 36.9 ns per loop (mean ± std. dev. of 7 runs, 1000000 loops each)

>%timeit sparse[500:1000]
55.9 ms ± 7.75 ms per loop (mean ± std. dev. of 7 runs, 10 loops each)

The sparse array is more efficient in memory, but the access time for multiple values is almost constant for dense arrays, but scales badly for the sparse version. So I don't think thats a viable alternative.

WinfriedL, 2018-04-11:
I think the first comparisons are not completly fair, since in my usecase the sparsearray would have less entries than the dense array. But from my own tests the performace advantage of the dense array still holds.

One might do a bit better than a dict  by using a pandas series, where the index is the pandapower index and the value is the ppc index. But still not as fast as your array implementation.

I have added a check/warning for the generator index analog to the exisiting check for the bus index in the pull request, this might be usefull in the future.