e2nIEE/pandapower #94: Allow consideration of dielectric losses in transmission lines

WinfriedL, 2018-03-28:
The π-equivalent circuit for lines currently only considers shunt susceptances at both ends, but in principle they have a shunt conductance as well. Those are needed to calculate dielectric losses in the line.

I tricked pandapower/pypower to consider this by setting `c_nf_per_km` to a complex value, where the imaginary part is `Im(c_nf_per_km)= - conductance / 2 pi f`.

However, I have to set the dtype of the column in the line dataframe to 'complex128':
`net.line.c_nf_per_km=net.line.c_nf_per_km.astype('complex128')`

Would it be possible to implement this directly in pandapower? I think you are doing a similar trick already for transformers to consider ferromagnetic losses:

https://github.com/lthurner/pandapower/blob/aef97f1f16ab496dd7d0b772c486b6e3be566b93/pandapower/build_branch.py#L195-L208

One would have to implement an additional parameter for lines, for example `p_diel_kw`, which is the parameter I have given for my case.

Another possibility would be to allow for complex values in the first place and maybe mention this trick in the documentation?

lthurner, 2018-04-03:
Yes as you pointed out we already do this for iron losses in transformers, it wouldn't be a problem to also implement this for lines. The only reason its not implemented yet is that we never had a use case where we needed this.

Using a complex number is a valid workaround for now, but introducing a new parameter is definitely cleaner and more consistent with the rest of pandapower. You proposed p_diel_kw, but in any case it should probably be p_diel_kw_per_km, right? An alternative would be to define it as a conductance value? Would be more in line with the other parameters, which are defined as reactance, resistance etc. Or is the loss as active power more common as a parameter?

WinfriedL, 2018-04-03:
You are right, it should have been p_diel_kw_per_km.  I don't know which is more common, I proposed p_diel because the software/data I have access to use it. A conductance value would be fine too.

lthurner, 2018-04-03:
I pushed a first implementation of the dielectric losses: 

- I introduced the new parameter is g_us_per_km for the conductance value in micro Siemens. I researched a little about the parameter, and found it most often defined as a conductance value rather than an active power value.
- the create functions and convert_format functions are adapted so that g_us_per_km can be specified, but always defaults to zero if it is not given. It can also be specified as a standard type parameter for lines.
- I adapted the results test that tests values against PowerFactory, and the voltage and active power values match exactly with PowerFactory
- included the parameter in the documentation for lines

Please check out if this works for you as expected.

WinfriedL, 2018-04-03:
Wow, thanks!
I will try it out as soon as I can (next week, since I am on vacation at the moment...)
Just wondering if the explanation in the documentation is clear for all users, especially since it is an often neglected parameter. "conductance of the line" could be interpreted just as the inverse of the resistance of the line?

lthurner, 2018-04-04:
Yes thats true, only conductance is probably not clear enough. I changed it to "dielectric conductance".

WinfriedL, 2018-04-10:
I have tested the implementation, it works as expected, thanks. 
The results are the same as in my hack and match the expected values.
I have not tested it as a standard type parameter since I am not using standard types.

lthurner, 2018-04-10:
Great then I am closing this issue now. With regard to the standard types, I also wrote a [test](https://github.com/lthurner/pandapower/blob/develop/pandapower/test/api/test_create.py#L123) to ensure it works correctly.

We will soon release a new version which includes this update.

closed 2018-04-10 by lthurner