e2nIEE/pandapower #99: Piecewise linear costs with uneven slopes

friederikemeier, 2018-04-12:
Currently, it is not possible to run an OPF with piecewise linear costs, where one slope is positive and one slope is negative as in the following example:

pp.create_piecewise_linear_cost(net, 0, "gen", np.array([ [-10, 8],[-6,0],[0,5]], "p")

It should be checked, what MATPOWER's behaviour in this case is and how we can fix this. I checked this about a year ago with MATPOWER for reactive power costs and it didn't converge in MATPOWER either. I don't know the current state of this bug in MATPOWER.

Workaround: I solved this using polynomial costs instead, so if it is possible, switch to a quadratic cost formulation. You can enforce the wanted behaviour by shifting the zero passage of a parabola (In the above case (x-6)^2 = x^2 - 12x + 36).

Help on this is highly appreciated! You can as well contact me under friederike.meier@uni-kassel.de.

tommasoorlandini, 2018-04-13:
Thank you for the post! One possible solution to avoid using piecewise linear costs with uneven slopes, could also be creating an additional generator for upward regulation (linear piecewise cost function) and an additional load for downward regulation, instead of using the generator for both upward and downward.

friederikemeier, 2018-05-22:
Hey @tommasoorlandini Did you get it to work like this? i would appreciate an update on this from your side! Kind regards!

tommasoorlandini, 2018-05-29:
Hey @friederikemeier! Yes I avoided using uneven slopes by implementing an additional generator for up regulation and a load for down regulation, and for each of them I created a linear plynomial function for costs. The results seems fine! Regards

friederikemeier, 2018-05-30:
@tommasoorlandini  Thanks for your feedback! I will add this into our documentation in the known caveats. And I will test this for reactive power, too. I don't think that we will provide an automatic conversion of those uneven slopes into a load and a generator, as we want to keep the results consistent with pypower/matpower. I think it would be helpful to introduce a warning in this case.
Regards

closed 2019-03-14 by friederikemeier