e2nIEE/pandapower #100: separate OPF cost function for load and gen

FlorianShepherd, 2018-04-16:
Cost definition in OPF is confusing for elements, which have positive and negative power (ext_grid storage).
 
Suggestion: Write a cost creation function (additionally to linear and polynomial) with the parameters "cost_gen" and "cost_load", which are both positive. Internally a piecewise linear cost function is created from this.

FlorianShepherd, 2018-04-16:
How about this syntax:

create_gen_cost(cost_gen = 0.15, max_generation = 1000, min_generation = 3, override =True)
create_eg_cost(cost_gen = 0.15, max_generation = 1000, cost_load = 0.3, max_load= 500)
create_eg_cost(cost_gen = 0.15, max_generation = -net.gen.min_p_kw, cost_load = 0.3, max_load= 500)

Constraints should be None by default. If they are given, the element tables are overwritten, if override is True(Its True by default).
Catching negative values and catcing that max must be greater than min.

???

friederikemeier, 2018-11-12:
Maybe start discussing this again in terms of #209

lthurner, 2018-11-13:
This was only discussed as a workaround because of the negative cost definition. If we sort out the costs I don't think the toolbox functions will be needed any longer

friederikemeier, 2018-11-13:
Should we then close this one?

closed 2018-11-13 by lthurner