e2nIEE/pandapower #105: Initialisation of powerflow with results not possible

julianw00, 2018-05-14:
Since updating to vs 1.5.0 when using
`pandapower.runpp(net, init="results")`
I receive the error :
"ValueError: could not broadcast input array from shape (161) into shape (37)"

I believe this is a new bug introduced with the update. Have others also encountered that?

ascheidl, 2018-05-15:
it seems like you are the only one to encounter this problem, yet. (All our tests pass.)

Can you provide a minimal example ?

You could also send your network model to me, so I can check it.

julianw00, 2018-05-17:
I found the cause. The error occurs when the powerflow results do not correspond to the network topology, e.g. if a line was added after executing the powerflow.

It's a pity that the powerflow caluculation can apparently no longer be initialised like that when there are no corresponding result stored for added branches. Until vs. 1.4.3 it worked and was more efficient that starting the iteration with 1 pu at each bus.

The following code raises the error
```import pandapower as pp
import pandapower.networks
net = pp.networks.create_cigre_network_lv()
pp.runpp(net)
pp.create_line(net, 
               from_bus=2, 
               to_bus=3, 
               length_km=1, 
                         name="new line" +str(1),
               std_type="NAYY 4x150 SE")               
pp.runpp(net, init="results")

FlorianShepherd, 2018-05-17:
Thank you julianw00 for the bug report.

@ascheidl and me just discussed this issue. 
The problem occurs because the results are not emptied if init=="results" is set. We decided that just the power flow should be initialized with the results (e.g. the voltage vector), but the results should not be kept, like it is performed when init == "flat". 

I'll try to fix this today.

FlorianShepherd, 2018-05-17:
5842f1283c37ada1b7aa9ea9380d5721ef352fa5 should fix this issue

closed 2018-05-17 by FlorianShepherd