e2nIEE/pandapower #106: Opf can't converge on the build-in test cases

chwneo7, 2018-05-16:
Hi, I try running opf on the case24_ieee_rts, but it can't converge. Then I test other build-in nets, most test cases meet the same problem. But in pypower, they worked properly. So I checked the net's polynomial_cost. It seems that the polynomial_cost in pandapower is converted wrongly. For example, polynomial_cost of case24 in pandapower is as follows: 
p | 0 | gen | [[0.0, 0.13, 400.6849]]
p | 1 | gen | [[0.0, 0.13, 400.6849]]
p | 2 | gen | [[1.4142e-08, 0.0160811, 212.3076]...
But, in pypower it's
p | 0 | gen | [[0.0, 130.0, 400.6849]]
p | 1 | gen | [[0.0, 130.0, 400.6849]]
p | 2 | gen | [[0.014142, 16.0811, 212.3076]]...
I replaced the polynomial_cost with original values in pypower, but it's confusing that the problem still exists.

friederikemeier, 2018-05-16:
So you say that after you replace the values the OPF still doesn't converge? 
I'll have a look at this.

friederikemeier, 2018-05-16:
If I run an opf and check on the internal gencost array, it looks like this for the generators mentioned above:
-0.00000000e+00,  -1.30000000e+02,         -4.00684900e+02
-0.00000000e+00,  -1.30000000e+02,         -4.00684900e+02
 -1.41000000e-02,  -1.60811000e+01,         -2.12307600e+02

First thing I observe here is the switched sign in the internal ppci. This is a bug that I already fixed myself. So in order to get the same results as in pypower, you must now multiply your costs with -1. This will be fixed in the next release, I already worked on this.

Second thing I observe is that we internally multiply the cost values with 1e6, 1e3 and so on... It is in make_objective.py in line 191. This is because in pandapower, the costs are defined per kW and in pypower per MW. So this is nothing to worry about because it results in the same internal gencost array.

So the quick fix for the sign error is:

```
for i in net.polynomial_cost.index:
    net.polynomial_cost.c.at[i][0] *=-1
```
Like this you should be able to reproduce the same results as in pypower. However, it still doesn't converge, which is really strange. I checked on this in Matpower and it does converge there, too. I think we may have some other problem in the conversion as well. I will check on this tomorrow!

friederikemeier, 2018-05-17:
In addition to that, the pypower case file is not fully represented in pandapower. All active and reactive power constraints are not in there. They must have been created with an older version of the pypower converter. So if you convert the pypower case file with the pypower converter again, you should be good. I checked the pypower converter for the constraints and they are in there, you should be able to run an OPF like this.

I also saw that my last commit is now on github, so you don't need to correct the sign anymore. All changes to the OPF are illustrated here:
https://github.com/lthurner/pandapower/blob/develop/tutorials/opf_changes-may18.ipynb

However, someone should  convert all case files again, test the opf on all case files and check if it converges and if the results are the same. So no closing of the issue until this is fixed.

chwneo7, 2018-05-18:
Hi, I just use the converter for test cases of pypower like this:

```
import pypower.case24_ieee_rts as c24
import pandapower as pp
import pandapower.networks as pn
import pandapower.converter.pypower.from_ppc as converter_py
net = converter_py(c24.case24_ieee_rts())
pp.runopp(net, verbose=True)
```
Unfortunately, it still can't converge. Maybe there are other bugs except for the disappearing of active and reactive power constraints.

friederikemeier, 2018-06-25:
So, even if the gencost array is the same, the initial solution is not the same.
Most of the matpower/pypower cases already have the solution already in the mpc/ppc case and the optimization starts from there. The pandapower OPF until now only features a flat start.

What we need to do to fix this: 
1. Allow initialization from results for OPF, too
2. Export results from  mpc/ppc files in converter

lthurner, 2018-11-15:
Actually the problem was simply that sgens where wrongly added from the ppc["bus"] table as controllable elements instead of as static elements, which lead to the OPF not converging. I fixed this and added the given case as a test in #213

closed 2018-11-15 by lthurner