e2nIEE/pandapower #107: Parsing Line elements: converting Ohm to Ohm-per-km?

nickvandewiele, 2018-05-17:
Hi,

Thank you very much for this great library!

However, I have a question regarding the creation of Line objects in Pandapower. Line elements must be created by specifying the _length_ of the line, and properties like _line resistance_ that are expressed on a per-length basis (e.g. `r_ohm_per_km`).

What if the input information of the lines of my network is expressed in _extensive_ units, like resistance in `Ohm`, rather than `Ohm per km`? What solutions do you recommend to model the properties of such lines? How was this tackled in your (commercial) parsers of input formats for PSS/E for example?

Thanks!

ascheidl, 2018-05-18:
Hi,

easiest solution is to assume all lines have length 1 km.

We do this for conversion from e.g. MATPOWER.

Cheers,
Alex

nickvandewiele, 2018-05-18:
Hi Alex,

Thanks for your quick answer!

So, by setting the line length to 1 km and using resistance as resistance-per-km in pandapower, you make the assumption that each line has the same length, but that each line _type_ is different.

Another approach could be to assume that all the lines are of the same type, and calculate the length of each line based on the `resistance-per-km` of the chosen line type.

Do you see any (dis)advantages of your approach compared to the second approach? Do you think there are impacts on results, e.g. for OPF calculations?

Thanks!
Nick

ascheidl, 2018-05-18:
Internally pandapower just multiplies all *_per_km values with length_km - so for the load flow calculation there is no difference.

I am not really up to date with the different possibilities to generate network-plots in pandapower. Maybe the length of lines is taken into acount somewhere there.

cheers,
alex

nickvandewiele, 2018-05-18:
Thanks for the reply. That answers everything I wanted to understand.

Nick

closed 2018-05-18 by nickvandewiele