e2nIEE/pandapower #111: Harmonic analysis

windymel, 2018-06-19:
Hi team,

Great work on this application so far. I'm an experienced consultant in the field of PSA and I'm interested in using this tool, potentially for commercial applications. However, I note there is currently no support for Harmonics Analysis. Are there any plans to expand the tool to include this feature? I would be interested in contributing to such a project.

lthurner, 2018-06-25:
Hi, as far as I know there are no concrete plans to include harmonic analysis in pandapower right now, but we would definitely assist in implementing this if you and/or others would be willing to implement harmonic analysis features!

lthurner, 2019-02-19:
Closing for inactivity, if there is interest in this please reopen anytime

closed 2019-02-19 by lthurner