e2nIEE/pandapower #112: Size of Ybus

ManuelMBaumann, 2018-06-22:
`import numpy as np
import pandas as pd

import pandapower as pp
import pandapower.networks as pn
import pandapower.plotting as pnp
from pandapower.plotting.plotly import pf_res_plotly

from pypower.makeYbus import makeYbus
from pandapower import converter as ppconv

import matplotlib.pyplot as plt

try:
    import seaborn
    colors = seaborn.color_palette()
except:
    colors = ["b", "g", "r", "c", "y"]

### Load test case
#der_opt = 'all'       # powerpanda's test case with renewables & batteries
der_opt = False        # powerpanda's test case w/o renewables & batteries
#der_opt = 'pv_wind'   # powerpanda's test case with only renewables
ppc = pn.create_cigre_network_mv(with_der=der_opt)

### Convert pp network to pypower and spy-plot Ybus
net = ppconv.to_ppc(ppc)
baseMVA, bus, branch = net["baseMVA"], net["bus"], net["branch"]
Ybus, Yf, Yt = makeYbus(baseMVA, bus, branch)

plt.spy(Ybus)
plt.show()

### Check if Ybus has dimension n_bus times n_bus
n_bus = ppc.bus.shape[0]
assert(Ybus.shape[0] == Ybus.shape[1] == n_bus)    

### Solve power flow equations
pp.runpp(ppc, algorithm='nr', max_iteration=100)  # nr = Newton-Raphson

### Plot network results
lc = pnp.create_line_collection(ppc, ppc.line.index, color="grey", zorder=1)
bc = pnp.create_bus_collection(ppc, ppc.bus.index, size=80, color=colors[0], zorder=2)
pnp.draw_collections([lc, bc])
pf_res_plotly(ppc)
`

ManuelMBaumann, 2018-06-22:
Here you can see the result of a power flow eqn. solve:

[image]

ascheidl, 2018-06-22:
Hi,

in the network are three open line-bus switches, so pandapower internally creates three auxillary buses. See http://pandapower.readthedocs.io/en/v1.5.1/elements/switch.html#electric-model

Cheers,
Alex

closed 2018-06-25 by lthurner